OllisGit/OctoPrint-SpoolManager #85: Choose spool popup instead of confirm popup?

JelledFro, 2020-11-11:
Instead of having to scroll down and choose then confirm the spool it would be nice to have the choice of a popup prompting you to choose the spool when you click print. Other than that the plugin works great!

OllisGit, 2021-01-17:
Hi @JelledFro,

yes that was also my first idea how this feature should implemented....but then I decided to not invest time for implementing an additional dialog (every minute counts ;-)

I put this to my long term roadmap.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-01-24:
fyi: Changed to mid-term

OllisGit, 2022-01-07:
fyi: will be in after the next release and will look like this:
[image]

jarbinier, 2022-01-16:
Hi,do you have a touchUI preview ?

taintedkernel, 2022-01-30:
Looks great!  I was wondering if the popup would appear if printing is started from a slicer (using Cura here).  On my system, when I print from Cura, I do not see any dialog or confirmation appear so I have to make sure the right spool is selected in OctoPrint first.  But when I start a print from OctoPrint itself, the confirmation appears as selected.  I can open another issue on this if it helps.

OllisGit, 2022-01-31:
Hi @taintedkernel,

this is technical not possible (...or at the moment I have no idea how to do this). 
The current implementation "hijacks" the start-print event from the browser-button and then the "spool-check"is done.

The Slicers use the OP-API on the server-side and there is no way to intercept the "start-print-event"...it is just starting the print immediately.

I need to ask the community, maybe an other developer has an idea.

taintedkernel, 2022-02-01:
@OllisGit thanks for the info.  Judging from other behavior I noticed, I'm not surprised by this limitation.

I guess it might be possible to do via a slicer plugin?  But even if so, that would be quite a bit of additional work.

OllisGit, 2022-02-01:
fyi: I asked the community: https://community.octoprint.org/t/intercepting-start-print-api-call/41502

taintedkernel, 2022-02-03:
Thank you!  I'll follow that thread as well :smile:

closed 2022-03-16 by github-actions[bot]