OllisGit/OctoPrint-SpoolManager #86: Considerung, you just took over FilamentManager, will this project keep going?

codingcatgirl, 2020-11-30:
I've noticed that since you took over the FilamentManager plugin, there has been no activity on this Plugin. 

Are you planning to continue development on this plugin or is FilamentManager now the future again?

ModischFabrications, 2020-12-13:
As an addition, I would also appreciate a comparison. Both seem to do pretty similar things, which plugin is preferred for which use case?

codingcatgirl, 2020-12-13:
The FilamentManager's README seems to touch on this topic a bit, but does not completely answer it, fyi

ModischFabrications, 2021-01-12:
Still waiting for an answer from @OllisGit , sorry StaleBot

codingcatgirl, 2021-01-12:
Feels silly to have to repeat that all the time, but that unhelpfulness is the nature of stalebot.

OllisGit, 2021-01-17:
Hi @codingcatgirl,
see #75

Short: 
SpoolManager = First class citizen -> implementation of new features
FilamentManager = Second class citizen -> bug fixing, review of PRs, no "heavy" enhancements 

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.
PSS. Stale bot is now deactivated

closed 2021-01-24 by OllisGit