OllisGit/OctoPrint-SpoolManager #87: Automatic Calculation of remaining filament - Filament Manager vs./and Spool Manager?

OliPrint, 2020-11-30:
Dear Olli, dear All,
I am completely new to octoprint.
What I am looking for is the possibility to:

1: Save a certain spool / all my spools (brand, material, ...)

2: Linking a spool to a print job - and in the same situation receiving the information before the print, if the remaining filament on that spool is sufficient for this planned job.
And there I think the remaining length would be the most appropriate/helpful figure (see also 3)

3. After completion of a print job, the used lenght is automatically deducted from that spool

And now my struggle:
Maybe I am doing something wrong, or I havent understood the usage of the PlugIns in the correct way.
I have installed both Plug Ins - Filament Manager and the Spool Manager, but I am not able to fulfil the above stated functionality.

In addition I learnt, that they maybe will be merged together, and I want to avoid lost effort in creating data in one plugin, and then maybe later I have to do it ones again for the same prints.

Could you please help me with my above stated need.
How can I manage this in the correct way, and which Plugin is the correct one for this, or do I need both.

Many many thanks in advance for any help
Oliver

SM-26, 2020-12-08:
I've got this kind of setup, currently I'm only using Filament manager and not Spool Manager

at what point are you stuck?
(saying "it does not work" does not help us understand what isn't working)

OliPrint, 2020-12-09:
Dear SM-26,

I am using the **PintJobHistory**, and I have installed both PlugIns - **Filament Manager** and the **Spool Manager**. 

Remark: As far as I learned from other postings, the spool manager is now in new hands and will most take over the filament manager.

1. And non of them is automatically asking me at the start of a print Job to choose the respective spool or filament? But I have marked in both cases for a reminder
2. Therefore there could be no automatic reduction regarding the used filament (which is my understanding of one of the funtionalities of the PlugIns - or do I misunderstand something here?)
3. After completion of the print job, I have to type in which filament I used, which is asked and stored in the PrintJobHistory
4. Based on the above there is no pre-selection of spool, and I do not realy see anywhere the automatic calculation what is remaining of the respective spool (weight or Lenght)

I would be very happy for any help
Thx

OliPrint, 2020-12-09:
Dear SM-26,
I have now disabled the spool manager, And I am running only the filament plugin.
And now, it works!

There is only one question remaining for me: in the PluginColumn/Filament Manager window is now stated Tool 0: ... including the remaining weight, which is already great info. Is there a possibility, to show here in the Tool = select box the REMAINING LENGTH instead?

Many thanks in advance
Oliver

SM-26, 2020-12-09:
As far as I know, nope.
That would be a new feature request that @OllisGit would need to consider.

But from time to time I've been using this:
https://jscalc.io/calc/XtRSYceWQbPpgcXW

The math under the hood is rather simple:
length: weight/(density*π*diameter^2/4)
when PLA value for density is about 1.25 and 1.75 for diameter

OliPrint, 2020-12-09:
Dear SM-26,
Thanks for the info. As far as I have seen the math seems already existing in the Filament Profile Page.
But I will try to get in contact with @OllisGit
Greetings

OllisGit, 2021-01-17:
Hi @OliPrint,
the Spoolmanager should do the job after the print and substract the right amount of filament from the spool. If not, then it is a bug.
Regarding the other mentioned topic....I think you should wait till the integration of both plugins (PJH and SpoolManager) is done..

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2022-01-07:
related to PR #206

OllisGit, 2022-01-12:
Hi! 
The mentioned PR is now in the latest release!

Because I have a lot of issues related the weight calculation I close this issue to get a more "up to date overview" in the issue-tracker.
If you found an issue in the latest version feel free to raise a new issue!

closed 2022-01-12 by OllisGit