OllisGit/OctoPrint-SpoolManager #88: Error in Log file (not sure if any behavior is changed)

pramodbutte, 2020-12-14:
I am getting following error in my log files:

I am not sure how this is affecting the behavior yet.

2020-12-13 22:45:04,535 - octoprint.plugins.DisplayLayerProgress - INFO - File select-event processing done!'
2020-12-13 22:45:04,546 - octoprint.plugin - ERROR - Error while calling plugin SpoolManager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_SpoolManager/__init__.py", line 401, in on_event
    self._on_file_selectionChanged(payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_SpoolManager/__init__.py", line 349, in _on_file_selectionChanged
    allFilemants = metadata["analysis"]["filament"]
KeyError: 'filament'
2020-12-13 22:45:04,558 - octoprint.plugins.DisplayLayerProgress - INFO - Store layer count in MetaFile
2020-12-13 22:45:04,671 - octoprint.plugins.DisplayLayerProgress - INFO - Read total height from MetaFile

Anyone seeing the same error? I have the latest octoprint and plugins (all up to date) as of Dec 13th 2020, 4:01 pm PST

thanks.

OllisGit, 2021-01-17:
PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-01-24:
Hi @pramodbutte,
fixed is in latest release v1.3.0

BR
Olli

closed 2021-01-24 by OllisGit