OllisGit/OctoPrint-SpoolManager #90: Improvement idea: Database of spools

wernight, 2020-12-14:
I'd be happy to contribute to a database of spools. Especially their ideal temperature and empty spool weight are important.

Simplest solution: Have them on GitHub in a sub-repo and have a link from the import/export page to the README of that GitHub. Then people can send pull requests.

Once that repo is filled a bit, the plugin could autofetch from that database as well.

OllisGit, 2021-01-17:
Hi @wernight,

good idea and I like that you say "Simplest solution".

I see a lot of challenges (I am not using the word "problems"):
- What informations should be collected in detail? Name every field
- Who and how should a user be add/modify the Spooldatabase-Repo?
- How should the GitHub Spool-Knowledgedatabase be structured? JSON, CSV, XML

- After the database is setup, should the plugin do a 1:1 sync, so that every repo spool is also in the plugins database-plugin...or instead using a database. just switch to github as the storage-provider
- ....

Don't get me wrong...i like the idea! But before implementing such feature there is a lot of conceptional work needed (user-workflow, UI,, usecase descriptions, etc.)
If your willing to make a concept that covers/answers a lot of questions I can put this feature to my roadmap.

An other option could be to include an already existing database/directory: https://www.filaments.directory/ ....if they have a technical interface to grab the needed data.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-04-15:
No feedback. Closed, because I need to cleanup the issue-tracker.
If needed, raise a new ticket.

closed 2021-04-15 by OllisGit