OllisGit/OctoPrint-SpoolManager #93: Doesn't work with Octolapse

deejay2, 2021-01-09:
Hi!

It looks like enabling the Octolapse plugin disables SpoolManager capability to account for filament use in prints... :(

OllisGit, 2021-01-17:
Hi @deejay2,

did you also raise the issue in the octolap issue tracker (if yes, please add the url here. if not, please do).
Or raise a question in the Octoprint Forum (https://community.octoprint.org/)..maybe someone has an idea. 

Unfortunately I don't use Octolaps so it would be very hard to analyze this issue.

deejay2, 2021-01-19:
Issue opened on Octolapse: https://github.com/FormerLurker/Octolapse/issues/671

OllisGit, 2021-01-24:
Hi @deejay2,

are you printing with a multi-extruder printer? Currently SpoolManager only supports a single tool....but you said it works when Octolaps is disabled.

Can you attach a sample gcode to this issue? Maybe I can identify some issues in SpoolManager.

FormerLurker, 2021-02-25:
@OllisGit, I haven't had a chance to look at this yet, but I plan to.  I have no idea how Octolapse could be messing with your plugin though :(  I'm sure it will be evident.  

I don't use Spool Manger, but I have no problem installing it.  Should be relatively easy to test using 'Test Mode' or 'Dry Run' :)

OllisGit, 2021-04-15:
No feedback. if error occurs again, create a new ticket.

Closed, because I need to cleanup the issue-tracker.

closed 2021-04-15 by OllisGit