OllisGit/OctoPrint-SpoolManager #96: HX711 and Load Cell

parskofamily, 2021-01-17:
Hi guys,

I just installed an HX711 onto my printer, and was able to get the [filament_scale](https://github.com/dieki-n/Octoprint-Filament-Scale) plugin to work last night.  I had to modify the code to work with python 3, and managed to do so.

I had a thought of combining it with your spoolManager.  I reviewed some code, and seems to be an easy thing.  Not sure if it has been requested, or if possible.  I am now up to 10 spools in my inventory (not as much as most it seems).  I would love to use my load cell to keep the weight up to date versus parsing Gcode.  Seems way easier, and is only two pins on the Pi.

Thoughts?

-Parsko

OllisGit, 2021-01-17:
Hi @parskofamily,

I like the idea, (already tracked in #4)

I also have all the needed hardware lying on my desk (loadcell, esp32, ...), but I have no time to work on it.
It looks like you are also a python developer....maybe you can work on it and make an PR.

What do you think?

BR Olli

parskofamily, 2021-01-18:
I was kinda hoping you'd ask.  I've been using Octoprint now for a few years.   I've been writing Python now for nearly the same time.  I've been wanting to automate more stuff in Octoprint (an independent Z home/level process is one I've been thinking about for a while).  So, I'm motivated.

My primary issue is that I don't really know how to manage Github, forking, etc...  I am also not familiar with the structure of a plugin (yet).  I _am_ becoming more familiar after having made some changes to the filament_scale code last night.  And, I've been looking over your code in Github today.  I am also not familiar with Java.  I can tweak existing code, but no experience to generate code.  As such, I may need some advice on that front.

In theory, this "should be as easy" as removing the "parsing" portion of your code, and replacing the `_calculate_weight` on line 198 of __init__.py with something that points to a read of the load cell.  Maybe a reverse calculation to get remaining length?  Not sure yet.  I'm assuming all the rest could be left the same?

I guess I need to fork it and figure out how to manage that stuff.  Been wanting to figure that out for work reasons to.  I use Python to automate test fixtures and stitch together hardware that doesn't normally want to be stuck together.  At some point, I need to start using GIT code management.

Thoughts?

-Parsko

PS - I've been playing around with the spool manager today, I like it.  What is newer, filament manager or spool manager?  Are they both current?  They seem identical?

OllisGit, 2021-02-02:
There is good tutorial about developing OctoPrint-Plugins:
https://docs.octoprint.org/en/master/plugins/gettingstarted.html

or I made a kind of scratch-pad about some OP things (Setup PyCharm-IDE)...https://github.com/OllisGit/OctoPrint-KnowledgeBase/wiki/Setup-IDE

Major difference between Spool - and Filamentmanager
Fila: Option for external Database, MultiTool-Support
Spool: More fields for a spool, better UI, ( no external and no multitool, yet )

closed 2021-05-23 by OllisGit