OllisGit/OctoPrint-SpoolManager #99: Recalculate remaining filament

poulanker, 2021-01-20:
Hi @OllisGit 

I absolutely love this plugin, very nice work you have done.

However, sometimes I fail :-/ I forget to select the spool, before starting a print using another filament.

My guess is that others make the same mistake and have to recalculate the remaining filament on **both** spools, because both now shows wrong values for _Used Weight_ and _Remaining weight_.

You then have to weigh each spool with filament and calculate the value you must enter into the field _Used weight_ in order to get the correct value displayed in _Remaining weight_. 
This is both tedious and errorprone.

Of course one could just enter all the values in a spreadsheet and get the values, but I do think that it would be nicer to be able to click a button e.g. **Adjust remaing weight** where I can enter the measured weight of the spool with remaining filament. When pressing an OK button the _Used weight_ should then be automatically updated (used_weight = total_weight + spool_weight - measured_weight). The end result would then be that the _Remaining weight_ shows the correct value. 

btw I was a little confused regarding the field _Total weight_ when I first started using the plugin. I assumed that this was the weight of the filament **and** spool. But this led to a wrong value for _Remaining weight_ which is eventually displayed in the sidebar.

Best regards
Poul Anker

Sembazuru, 2021-02-10:
Nearly identical to the suggestion I made in the 4th comment of my issue over on https://github.com/OllisGit/OctoPrint-SpoolManager/issues/119
Are we building some sort of user consensus here? ;-)

poulanker, 2021-02-10:
@Sembazuru I guess we are ;-)

OllisGit, 2022-01-12:
Hi ! 
The mentioned PR is now in the latest release!

Because I have a lot of issues related the weight calculation I close this issue to get a more "up to date overview" in the issue-tracker.
If you found an issue in the latest version feel free to raise a new issue!

closed 2022-01-12 by OllisGit