OllisGit/OctoPrint-SpoolManager #103: Unable to update from 1.2.0 to 1.3.0

f-hnt, 2021-01-25:
The plugin cannot update from version 1.2.0 to 1.3.0. I've tried through the Octoprint environment and also manually. 
Also tried by uninstalling 1.2.0 and again I'm unable to install 1.3.0.

**Octoprint log:** 
Updating, please wait.+++++++++++++++++++++++++++++++++++++++++Now updating SpoolManager Plugin to 1.3.0+++++++++++++++++++++++++++++++++++++++++/home/hnt/OctoPrint/venv/bin/python -m pip --disable-pip-version-check install https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip --no-cache-dirCollecting https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zipDownloading https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip (4.9 MB)Requirement already satisfied: OctoPrint in /home/hnt/OctoPrint/venv/lib/python3.8/site-packages (from OctoPrint-SpoolManager==1.3.0) (1.5.3)Requirement already satisfied: peewee in /home/hnt/OctoPrint/venv/lib/python3.8/site-packages (from OctoPrint-SpoolManager==1.3.0) (3.13.3)Requirement already satisfied: pillow>=6.2.0<7.0.0 in /home/hnt/OctoPrint/venv/lib/python3.8/site-packages (from OctoPrint-SpoolManager==1.3.0) (7.2.0)Collecting psycopg2-binaryDownloading psycopg2-binary-2.8.6.tar.gz (384 kB)ERROR: Command errored out with exit status 1:command: /home/hnt/OctoPrint/venv/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-109stj67/psycopg2-binary/setup.py'"'"'; __file__='"'"'/tmp/pip-install-109stj67/psycopg2-binary/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-109stj67/psycopg2-binary/pip-egg-infocwd: /tmp/pip-install-109stj67/psycopg2-binary/Complete output (23 lines):running egg_infocreating /tmp/pip-install-109stj67/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-infowriting /tmp/pip-install-109stj67/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/PKG-INFOwriting dependency_links to /tmp/pip-install-109stj67/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/dependency_links.txtwriting top-level names to /tmp/pip-install-109stj67/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/top_level.txtwriting manifest file '/tmp/pip-install-109stj67/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/SOURCES.txt'Error: pg_config executable not found.pg_config is required to build psycopg2 from source.  Please add the directorycontaining pg_config to the $PATH or specify the full executable path with theoption:python setup.py build_ext --pg-config /path/to/pg_config build ...or with the pg_config option in 'setup.cfg'.If you prefer to avoid building psycopg2 from source, please install the PyPI'psycopg2-binary' package instead.For further information please check the 'doc/src/install.rst' file (also at<https://www.psycopg.org/docs/install.html>).----------------------------------------ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

**Manual install log:**
hnt@usrpi4:~$ /home/hnt/OctoPrint/venv/bin/pip install "https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip"
Collecting https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip
  Downloading https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip (4.9 MB)
     |████████████████████████████████| 4.9 MB 1.0 MB/s
Requirement already satisfied: OctoPrint in ./OctoPrint/venv/lib/python3.8/site-packages (from OctoPrint-SpoolManager==1.3.0) (1.5.3)
Requirement already satisfied: peewee in ./OctoPrint/venv/lib/python3.8/site-packages (from OctoPrint-SpoolManager==1.3.0) (3.13.3)
Requirement already satisfied: pillow>=6.2.0<7.0.0 in ./OctoPrint/venv/lib/python3.8/site-packages (from OctoPrint-SpoolManager==1.3.0) (7.2.0)
Collecting psycopg2-binary
  Using cached psycopg2-binary-2.8.6.tar.gz (384 kB)
    ERROR: Command errored out with exit status 1:
     command: /home/hnt/OctoPrint/venv/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jpmbtgf7/psycopg2-binary/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jpmbtgf7/psycopg2-binary/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-jpmbtgf7/psycopg2-binary/pip-egg-info
         cwd: /tmp/pip-install-jpmbtgf7/psycopg2-binary/
    Complete output (23 lines):
    running egg_info
    creating /tmp/pip-install-jpmbtgf7/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info
    writing /tmp/pip-install-jpmbtgf7/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/PKG-INFO
    writing dependency_links to /tmp/pip-install-jpmbtgf7/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/dependency_links.txt
    writing top-level names to /tmp/pip-install-jpmbtgf7/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/top_level.txt
    writing manifest file '/tmp/pip-install-jpmbtgf7/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/SOURCES.txt'

    Error: pg_config executable not found.

    pg_config is required to build psycopg2 from source.  Please add the directory
    containing pg_config to the $PATH or specify the full executable path with the
    option:

        python setup.py build_ext --pg-config /path/to/pg_config build ...

    or with the pg_config option in 'setup.cfg'.

    If you prefer to avoid building psycopg2 from source, please install the PyPI
    'psycopg2-binary' package instead.

    For further information please check the 'doc/src/install.rst' file (also at
    <https://www.psycopg.org/docs/install.html>).

    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

OllisGit, 2021-01-25:
Hi @f-hnt!

Damn,
sometimes the python environment is not able to install the postgres database driver... no idea why.
See:
- https://github.com/OllisGit/OctoPrint-FilamentManager/issues?q=is%3Aissue+psycopg2
- 

please try. to install ``psycopg2``` first and then the plugin

```
pip install psycopg2
```

Or you wait, till I setup a version without the db-drivers...sorry!

f-hnt, 2021-01-25:
> Hi @f-hnt!
>
> Damn,
> sometimes the python environment is not able to install the postgres database driver... no idea why.
> See:
>
>     * [...]

Thanks for the fast response, OllisGit.
I will wait for the version without the db-drivers.

Regards :)

OllisGit, 2021-01-25:
Hi @f-hnt,

new version is out, please test and give feedback.

Thx, in advance
Olli

VibroAxe, 2021-01-25:
Your new version forgot the master.zip so it's broken updates for all..

As a side note, given the shortness between 1.3.0 and 1.3.1 i'd suggest having the full 1.3.0 changelog on the release notes for 1.3.1 else the octoprint ui doesn't inform users about the changes!

VibroAxe, 2021-01-25:
@OllisGit 

I tried to install from the source.zip (which I believe is largely the same as Master.zip and i'm getting DB errors from SpoolManager

```

2021-01-25 10:24:00,285 - octoprint.plugins.eeprom_marlin - INFO - Finished data collection, updating UI
2021-01-25 10:24:05,179 - octoprint.server.util.flask - INFO - Passively logging in user vibs from 10.10.1.24
2021-01-25 10:24:05,180 - octoprint.access.users - INFO - Logged in user: vibs
2021-01-25 10:24:58,882 - octoprint.server.preemptive_cache - INFO - ... done in 63.42s
2021-01-25 10:24:59,354 - octoprint.server.util.flask - INFO - Passively logging in user vibs from 10.10.1.24
2021-01-25 10:24:59,356 - octoprint.access.users - INFO - Logged in user: vibs
2021-01-25 10:24:59,935 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.10.1.24
2021-01-25 10:25:02,075 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-01-25 10:25:02,076 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'password': 'illO', 'name': 'SpoolDatabase', 'host': 'localhost', 'fileLocation': u'/home/pi/.octoprint/data/SpoolManager/spoolmanager.db', 'user': 'Olli', 'useExternal': False, 'type': 'sqlite', 'port': 5432, 'baseFolder': u'/home/pi/.octoprint/data/SpoolManager'}
2021-01-25 10:25:02,079 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-01-25 10:25:02,081 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-01-25 10:25:02,082 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'password': 'illO', 'name': 'SpoolDatabase', 'host': 'localhost', 'fileLocation': u'/home/pi/.octoprint/data/SpoolManager/spoolmanager.db', 'user': 'Olli', 'useExternal': False, 'type': 'sqlite', 'port': 5432, 'baseFolder': u'/home/pi/.octoprint/data/SpoolManager'}
2021-01-25 10:25:02,085 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-01-25 10:25:02,104 - octoprint - ERROR - Exception on /plugin/SpoolManager/loadSpoolsByQuery [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_SpoolManager/api/SpoolManagerAPI.py", line 628, in loadAllSpools
    allSpoolsAsDict = Transformer.transformAllSpoolModelsToDict(allSpools)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_SpoolManager/api/Transformer.py", line 94, in transformAllSpoolModelsToDict
    for job in allSpoolModels:
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 6797, in __iter__
    self.execute()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 1886, in inner
    return method(self, database, *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 1957, in execute
    return self._execute(database)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2129, in _execute
    cursor = database.execute(self)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 3112, in execute
    return self.execute_sql(sql, params, commit=commit)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 3106, in execute_sql
    self.commit()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2873, in __exit__
    reraise(new_type, new_type(exc_value, *exc_args), traceback)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 3099, in execute_sql
    cursor.execute(sql, params or ())
OperationalError: no such column: t1.updated
2021-01-25 10:25:02,119 - tornado.access - ERROR - 500 GET /plugin/SpoolManager/loadSpoolsByQuery?from=0&to=10&sortColumn=displayName&sortOrder=desc&filterName=all (10.10.1.24) 55.31ms
2021-01-25 10:25:03,121 - octoprint.server.util.sockjs - INFO - New connection from client: 10.10.1.24
2021-01-25 10:25:05,399 - octoprint.server.util.flask - INFO - Passively logging in user vibs from 10.10.1.24
2021-01-25 10:25:05,401 - octoprint.access.users - INFO - Logged in user: vibs
2021-01-25 10:25:08,543 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-01-25 10:25:08,545 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'password': 'illO', 'name': 'SpoolDatabase', 'host': 'localhost', 'fileLocation': u'/home/pi/.octoprint/data/SpoolManager/spoolmanager.db', 'user': 'Olli', 'useExternal': False, 'type': 'sqlite', 'port': 5432, 'baseFolder': u'/home/pi/.octoprint/data/SpoolManager'}
[... cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2873, in __exit__
    reraise(new_type, new_type(exc_value, *exc_args), traceback)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 3099, in execute_sql
    cursor.execute(sql, params or ())
OperationalError: no such column: t1.updated
2021-01-25 10:25:13,591 - tornado.access - ERROR - 500 GET /plugin/SpoolManager/loadSpoolsByQuery?filterName=hideEmptySpools%2ChideInactiveSpools&from=0&to=3000&sortColumn=lastUse&sortOrder=desc (10.10.1.24) 51.66ms

```

VibroAxe, 2021-01-25:
I get the same issue in the settings page

[image]

OllisGit, 2021-01-25:
Please attach the full log-file, because after installation there should be a database-upgrade tasks executed.

```
Log: We need to upgrade the database scheme from: 3 to 4
```

VibroAxe, 2021-01-25:
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5866189/octoprint.log)

VibroAxe, 2021-01-25:
Wait, i might have done a dumb

f-hnt, 2021-01-25:
> 
> 
> Hi @f-hnt,
> 
> new version is out, please test and give feedback.
> 
> Thx, in advance
> Olli

Hi Olli, 

I've performed the update by the link below through the Octoprint and it went without errors. 
https://github.com/OllisGit/OctoPrint-SpoolManager/releases/download/1.3.1/master.zip

All existing spools are present and I have no such error on the Storage tab like the one, reported by VibroAxe.
[image]

For me everything looks fine for now. 

Sincere thanks :)

VibroAxe, 2021-01-25:
@OllisGit 

Looking from the command line, it appears to be failing the enclosureTemperature rename
```
sqlite> PRAGMA foreign_keys=off;
sqlite> BEGIN TRANSACTION;
sqlite> ALTER TABLE 'spo_pluginmetadatamodel' ADD 'updated' DATETIME;
sqlite> ALTER TABLE 'spo_pluginmetadatamodel' ADD 'version' INTEGER;
sqlite> ALTER TABLE 'spo_pluginmetadatamodel' ADD 'originator' CHAR(60);
sqlite> UPDATE 'spo_pluginmetadatamodel' SET version=1;
sqlite> ALTER TABLE 'spo_spoolmodel' ADD 'updated' DATETIME;
sqlite> ALTER TABLE 'spo_spoolmodel' ADD 'originator' CHAR(60);
sqlite> ALTER TABLE 'spo_spoolmodel' ADD 'materialCharacteristic' VARCHAR(255);
sqlite> ALTER TABLE 'spo_spoolmodel' ADD 'isActive' INTEGER;
sqlite> UPDATE 'spo_spoolmodel' SET isActive=1;
sqlite> CREATE INDEX spoolmodel_materialCharacteristic ON spo_spoolmodel (materialCharacteristic);
sqlite> CREATE INDEX spoolmodel_material ON spo_spoolmodel (material);
sqlite> CREATE INDEX spoolmodel_vendor ON spo_spoolmodel (vendor);
sqlite> ALTER TABLE spo_spoolmodel RENAME COLUMN encloserTemperature to enclosureTemperature;
Error: near "COLUMN": syntax error
```

VibroAxe, 2021-01-25:
Further the above

```pi@octopi:~/.octoprint/data/octoprint_SpoolManager $ sqlite3 spoolmanager.db
SQLite version 3.16.2 2017-01-06 16:32:41
Enter ".help" for usage hints.
sqlite> select sqlite_version();
3.16.2```

This is an older pi than I thought... brb dist-upgrading

OllisGit, 2021-01-25:
Hi @VibroAxe,

thx for analysing the issue...I found an google entry:
```
SQLite did not support the ALTER TABLE RENAME COLUMN syntax before version 3.25.0.
 ```
S, I need to find an other way to rename the misspelled field....give me a second (or maybe more then a second)

OllisGit, 2021-01-25:
Hi,

please try the new release 1.3.2. This version did not use the column rename function.

BR
Olli

VibroAxe, 2021-01-26:
@OllisGit sorry, I'd already upgraded to python 3.7 which included an update to debian buster. On the plus side my DB then upgraded correctly but I can't roll back to test this regresssion :(

OllisGit, 2021-02-02:
Hi @VibroAxe,
did you test the latest release 1.3.3?

VibroAxe, 2021-02-03:
@OllisGit as above, I'm running 1.3.3 quite happily, however I'd already upgrade to 3.7 with an up to date sqlite so I can't fully confirm it's working as intended. However, technically octoprint isn't support on that old a debian system, so I think you are probably ok to close this out

closed 2021-04-15 by OllisGit