OllisGit/OctoPrint-SpoolManager #104: Wrong database call after update to 1.3.0

v-abt, 2021-01-25:
Hi,

after updating to the latest release, my filament spools won't show up and octoprint is complaining about a wrong database call.
Here is the corresponding section of the octoprint log:
```
2021-01-25 10:09:45,492 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-01-25 10:09:45,493 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/octopi/.octoprint/data/SpoolManager', 'fileLocation': '/home/octopi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-01-25 10:09:45,496 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-01-25 10:09:45,496 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-01-25 10:09:45,497 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/octopi/.octoprint/data/SpoolManager', 'fileLocation': '/home/octopi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-01-25 10:09:45,501 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-01-25 10:09:45,528 - octoprint.plugins.SpoolManager.DatabaseManager - ERROR - Database call error in methode loadSpool
Traceback (most recent call last):
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 3129, in execute_sql
    cursor.execute(sql, params or ())
sqlite3.OperationalError: no such column: t1.updated

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_SpoolManager/DatabaseManager.py", line 503, in _handleReusableConnection
    return databaseCallMethode()
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_SpoolManager/DatabaseManager.py", line 601, in databaseCallMethode
    return SpoolModel.get_by_id(databaseId)
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 6441, in get_by_id
    return cls.get(cls._meta.primary_key == pk)
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 6430, in get
    return sq.get()
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 6860, in get
    return clone.execute(database)[0]
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 1898, in inner
    return method(self, database, *args, **kwargs)
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 1969, in execute
    return self._execute(database)
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 2141, in _execute
    cursor = database.execute(self)
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 3142, in execute
    return self.execute_sql(sql, params, commit=commit)
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 3136, in execute_sql
    self.commit()
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 2902, in __exit__
    reraise(new_type, new_type(exc_value, *exc_args), traceback)
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 185, in reraise
    raise value.with_traceback(tb)
  File "/home/octopi/OctoPrint/venv/lib/python3.7/site-packages/peewee.py", line 3129, in execute_sql
    cursor.execute(sql, params or ())
peewee.OperationalError: no such column: t1.updated
2021-01-25 10:09:45,528 - octoprint.plugins.SpoolManager - WARNING - SendToClient: error#DatabaseManager#Database call error in methode loadSpool. See OctoPrint.log for details!
2021-01-25 10:09:45,545 - octoprint.plugins.SpoolManager - WARNING - Last selected Spool from plugin-settings not found in database. Maybe deleted in the meantime.
```
I would appreciate any help!

v-abt, 2021-01-25:
Hi @OllisGit,
thank you for looking into this. Here is the log as a whole: 
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5865971/octoprint.log)

OllisGit, 2021-01-25:
Thanks for the log.

I tested different python versions, also your version 3.7.3, but I could not reproduce the error on my side. Database Upgrade and also the creation of a new database works.

Please do me a favour and install the current source zip and not the release zip.
- Plugin-Manager, URL: https://github.com/OllisGit/OctoPrint-SpoolManager/archive/master.zip
- New version 1.3.1-dev

I only added a log-statement, maybe I can see any issue after you restarted the Server.

An other test could be to delete the current database and then a clean database is created automatically....lets try number one.

v-abt, 2021-01-25:
I went ahead and updated to the version of the master branch. The error still persists. Here is the log:
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5866827/octoprint.log)

Is there anyway to manually edit and import my database to the new db style?

v-abt, 2021-01-25:
deleting the database and re-adding my spool(s) with data from the old database does seem to work fine.
Feel free to close this issue any time.

scurfin, 2021-01-25:
I've got the same error. Deleting the old one also didn't work.
[octoprint (4).log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5867320/octoprint.4.log)

OllisGit, 2021-01-25:
Fyi: I found an google entry:

SQLite did not support the ALTER TABLE RENAME COLUMN syntax before version 3.25.0.
So, I need to find an other way to rename the misspelled field....give me a second (or maybe more then a second)

OllisGit, 2021-01-25:
Hi,

please try the new release 1.3.2. This version did not use the column rename function.

BR
Olli

scurfin, 2021-01-25:
I think the update of the database scheme was successful (scheme 4 and item count 16 displayed in settings)
but I can't see any spools and I got a red error on the side:
"ERROR: SPM:DatabaseManager
Database call error in methode loadSpool. See OctoPrint.log for details!"

Last lines of my logfile (after installation):
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5868897/octoprint.log)

nielseulink, 2021-01-26:
> . This version did not use the colu

Issue is still present in version 1.3.2

scurfin, 2021-01-26:
Update from 1.2.0 to 1.3.3 works just fine and the database seems to be converted accordingly. 

Thanks for the quick fix!

nielseulink, 2021-01-27:
> > Hi,
> > please try the new release 1.3.2. This version did not use the column rename function.
> > BR
> > Olli
> 
> Issue is still present in version 1.3.2

Has been fixed in 1.3.3
Thank you for the fix!