OllisGit/OctoPrint-SpoolManager #105: spoolmanager update failing

mrseeker, 2021-01-25:
From the octoprint log:

```
+++++++++++++++++++++++++++++++++++++++++
Now updating SpoolManager Plugin to 1.3.1
+++++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip --no-cache-dir
ERROR: HTTP error 404 while getting https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip
ERROR: Could not install requirement https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/OllisGit/OctoPrint-SpoolManager/releases/download/1.3.1/master.zip for URL https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
```

chatainsim, 2021-01-25:
Same for me, there is no master.zip for this release. yet?

VibroAxe, 2021-01-25:
@OllisGit you've forgotten to add the master.zip to the release!

OllisGit, 2021-01-25:
Thanks for reporting. 
There is something from with the buildsystem from https://travis-ci.com/
In the past it works always like charm...I am analysing the issue.

OllisGit, 2021-01-25:
Download of the latest version 1.3.2 works.
Need a validation step in the build-pipeline, see #106

closed 2021-01-25 by OllisGit