OllisGit/OctoPrint-SpoolManager #107: Mask log-messages

OllisGit, 2021-01-25:
Like passwords and urls

PowerWiesel, 2021-04-19:
Hey :smile: 
I was about to report that passwords are leaked in the logs and saw that you already have this ticket.
I don't use your plugin but I guess the Olli user is the default one? 
If everybody got the same user it's not that bad but if somebody changes the credentials and uploads the logs in a ticket or on the forum it wouldn't be so great.

OllisGit, 2022-01-07:
In next release the connection settings logs are only printed, if sql logging is enabled (only for debugging)

OllisGit, 2022-01-12:
Hi,
in the latest version the issue should be fixed. Only visible if SQL logging is enabled.
Please test and give a feedback.

Thx in advance
Olli

closed 2022-02-27 by github-actions[bot]