OllisGit/OctoPrint-SpoolManager #109: DB Error after update to 1.3.2

StevenK2104, 2021-01-25:
Getting this error now after I updated to 1.3.2

ERROR: SPM:DatabaseManager
Database call error in methode loadSpool. See OctoPrint.log for details!

Log file below

[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5868896/octoprint.log)
[image]

StanDH, 2021-01-25:
Identical Problem.

StevenK2104, 2021-01-25:
any idea on how long this will be till the issue is fixed ?

OllisGit, 2021-01-25:
Hi @StevenK2104 ,

the log messages of the db-migration looks good. Please do the following:

- login to your raspberry via ssh
- navigate to the database-location (found in plugin-settings)
- type ```sqlite3 spoolmanager.db```
- type ```select * from spo_spoolmodel```
- paste the result to this issue
- type ```select * from spo_pluginmetadatamodel```
- paste the result to this issue

btw. "Impossible is done immediately, miracles take longer"

OllisGit, 2021-01-25:
Hi,
fyi: found the issue, will be fixed in the next minutes!

vdibenedetto, 2021-01-25:
pi@octopi:~/.octoprint/data/SpoolManager $ sqlite3 spoolmanager.db
-bash: sqlite3: command not found

schwos, 2021-01-25:
Experience the same issue; as well sqlite3 isn't installed natively on the Raspberry PI install in the image. It was necessary to install sqlite3 on my image

sudo apt-get install sqlite3

OllisGit, 2021-01-25:
Error is fixed in lates release 1.3.3.
Please test and give feedback.

Sorry,
Olli

StevenK2104, 2021-01-25:
Issue now resolved.
Many thanks @OllisGit for the great work on this

vdibenedetto, 2021-01-25:
It works, thank you!

closed 2021-01-25 by StevenK2104