OllisGit/OctoPrint-SpoolManager #110: Lost DB after update to 1.3.2

PGlyre, 2021-01-25:
All my information in my DB is lost, 
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5868979/octoprint.log)

jsn0327, 2021-01-25:
Same here... The backup DB is present though.... The errors in my log file are the same as the issue that was posted for updating to 1.3.0, but my logs show that the DB was successfully migrated.

2021-01-25 12:02:52,246 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2021-01-25 12:02:52,331 - octoprint.plugins.SpoolManager - INFO - Start initializing
2021-01-25 12:02:52,337 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Init DatabaseManager
2021-01-25 12:02:52,339 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databasefile '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db' exists: True
2021-01-25 12:02:52,340 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-01-25 12:02:52,341 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'password': 'illO', 'name': 'SpoolDatabase', 'host': 'localhost', 'fileLocation': u'/home/pi/.octoprint/data/SpoolManager/s$
2021-01-25 12:02:52,343 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-01-25 12:02:52,344 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Check if database-scheme upgrade needed...
2021-01-25 12:02:52,346 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Current databasescheme: 3
2021-01-25 12:02:52,347 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - We need to upgrade the database scheme from: '3' to: '4'
2021-01-25 12:02:52,347 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Starting database backup
2021-01-25 12:02:52,350 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Backup of spoolmanager database created '/home/pi/.octoprint/data/SpoolManager/spoolmanager-backup-20210125-1202.db'
2021-01-25 12:02:52,351 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database migration from '3' to '4'
2021-01-25 12:02:52,352 - octoprint.plugins.SpoolManager.DatabaseManager - INFO -  Starting 3 -> 4
2021-01-25 12:02:52,507 - octoprint.plugins.SpoolManager.DatabaseManager - INFO -  Successfully 3 -> 4
2021-01-25 12:02:52,508 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - ...Database-scheme successfully upgraded.
2021-01-25 12:02:52,508 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database created-check done
2021-01-25 12:02:52,511 - octoprint.plugins.SpoolManager - INFO - Done initializing

My spools are empty though and I am receiving an error in OctoPrint: 

octoprint.plugins.SpoolManager.DatabaseManager - ERROR - Database call error in methode loadSpool

Here is my log:
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5869114/octoprint.log)

vdibenedetto, 2021-01-25:
The same for me:

2021-01-25 20:20:20,200 - octoprint.plugins.SpoolManager.DatabaseManager - ERROR - Database call error in methode loadSpool
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_SpoolManager/DatabaseManager.py", line 561, in _handleReusableConnection
    return databaseCallMethode()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_SpoolManager/DatabaseManager.py", line 659, in databaseCallMethode
    return SpoolModel.get_by_id(databaseId)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 6395, in get_by_id
    return cls.get(cls._meta.primary_key == pk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 6384, in get
    return sq.get()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 6807, in get
    return clone.execute(database)[0]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 1886, in inner
    return method(self, database, *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 1957, in execute
    return self._execute(database)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2129, in _execute
    cursor = database.execute(self)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 3112, in execute
    return self.execute_sql(sql, params, commit=commit)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 3106, in execute_sql
    self.commit()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2873, in __exit__
    reraise(new_type, new_type(exc_value, *exc_args), traceback)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 3099, in execute_sql
    cursor.execute(sql, params or ())
OperationalError: no such column: t1.updated

OllisGit, 2021-01-25:
Database is not lost, each time an upgrade is made a backup of the database is created.

Also please try the database-connection  as described here https://github.com/OllisGit/OctoPrint-SpoolManager/issues/109#issuecomment-767098829

This ticket is duplicated use this one #109.  
Closed!

closed 2021-01-25 by OllisGit