OllisGit/OctoPrint-SpoolManager #111: QR Code Link ignores Reverse Proxy "proxy_set_header X-Script-Name"

lightmaster, 2021-01-26:
My Octoprint server is behind a Reverse Proxy where the URL is in the form of `example.com/octoprint/`. SpoolManager's url is therefore: `example.com/octoprint/#tab_plugin_SpoolManager`. For the URL to print out the QR Code, SpoolManager is printing it out as just `example.com/plugin/SpoolManager`, dropping `/octoprint` from the url.

Just as a reference, the /octoprint is added by this in my nginx config:
```
proxy_set_header		X-Script-Name /octoprint;
```

lightmaster, 2021-01-26:
Oddly enough, the QR Code itself contains the `/octoprint` part of the URL, just the link to get to the QR Code does not contain it.

OllisGit, 2021-01-26:
Hi @lightmaster,

just: for my understanding
- the thumbnail-qr code is shown in your edit-dialog, right ?
- if you click on the thumbnail you receive a 404 (or page not found), because of the reverse-proxy
- if you patch the browser-url manuely you receive the full picture
- scanning the full picture includes the right url and the edit-spool page could be opened.

Short: the link from the thumbnail is broken, is this correct

lightmaster, 2021-01-26:
See my screenshots in #112, the QR Code is not loaded in the edit dialog, but there is a link to the picture. Everything else, yes.

VibroAxe, 2021-01-26:
@lightmaster if you go to example.com/octoprint/plugin/SpoolManager/generateQRCode/4 and scan the code, what url is it taking you to?

lightmaster, 2021-01-26:
`example.com/octoprint/plugin/SpoolManager/selectSpoolByQRCode/4`

VibroAxe, 2021-01-26:
@OllisGit this come from [L259-260](https://github.com/OllisGit/OctoPrint-SpoolManager/blob/4368360547a66bb0b05486a444b0fb001fed1c83/octoprint_SpoolManager/templates/SpoolManager_tab_dialogs.jinja2#L259`) in the dialogs jinja2. I'm not familiar enough with Jinja in Octoprint to know how to get the basepath for relative browsing

@lightmaster can you try installing from https://github.com/VibroAxe/OctoPrint-SpoolManager/archive/master.zip I think I have a solution

lightmaster, 2021-01-26:
@VibroAxe Yea, gimme a little. Updating OctoPi to the new version right now.

lightmaster, 2021-01-26:
@VibroAxe Yup, QR Code does work now with that fork. Annoyingly, there's still the other issue I submitted where SpoolManager and UI Customizer do not play nice together, so it brings up an edit dialog I can't scroll down to close on my phone.

VibroAxe, 2021-01-26:
@OllisGit we can close this if you are happy to accept #115

commit 0e86e1b3 (2021-01-27): Resolution to QR code scanning issue (#115)

* Update SpoolManager.js

Fix javascript QR code selection

* Update SpoolManager.js

Fix sidebar not being updated when spool selecting by QR Code

* Update SpoolManager.js

Display the SpoolManager tab when scanning a spoolManager barcode

* Update SpoolManager.js

* Update SpoolManager.js

Perform tab switch after selectSpool to allow page to finish loading

* Use relative pathing for QR Code

Fixes #111 using relative pathing,
It might be better to do this using jinja native but I can't currently find a basepath reference

* Update SpoolManager_ [... cut]
closed 2021-01-27 by OllisGit