OllisGit/OctoPrint-SpoolManager #112: Spool page is not 2 columns and missing note's text box

lightmaster, 2021-01-26:
The picture in the release notes of a Spool's page looks nothing like what the same page on my server looks like. Mine is one long page that has too be scrolled down a lot, there is no image of a QR Code, just a hyperlink to one, and the notes text box for mine is now missing.

Your picture from release section
[image]

My Picture from my server
[image]
[image]

OllisGit, 2021-01-26:
Hi @lightmaster,

which screen resolution and which browser do you use?
please try to zoom-out with the browser tool (Ctrl - or use the main menu or scroll mouse) 

Are you able to inspect the image link with the Dev-Tools from the browser or do you need guidance?
Take a look into the network tab, there should be a red line for the missing image, please copy this error here

lightmaster, 2021-01-26:
I'll do all that when I get back to my computer. I know resolution is 1920x1080 and browser is Firefox.

lightmaster, 2021-01-26:
Zooming out to 80% makes half the text box at the bottom appear, zooming out to 70% makes the other half appear. I do have a theme plugin installed, I'll try disabling that to see if its a conflict between SpoolManager and it.

The missing QR Code is that same as my other bug report, issue #111, `example.com/plugin/Spoolmanager/generageQRCode/4`, missing the `/octoprint` after `.com`.

lightmaster, 2021-01-26:
UI Customizer seems to be interfering with the edit dialog. Not getting the QR code or link right either way, but without UI Customizer, the edit dialog isn't 1 long weirdly spaced column.

VibroAxe, 2021-01-26:
@lightmaster if you go to `example.com/octoprint/plugin/SpoolManager/generateQRCode/4` and scan the code, what url is it taking you to?

LazeMSS, 2021-01-27:
Its not UI Customizer that destroying - its standard bootstrap responsive thats doing that. Having full (12) span inside a modal inside a row does that. It also happens when trying on a clean boostrap 2.3.2

@OllisGit it can be fixed by making the "row" on https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/octoprint_SpoolManager/templates/SpoolManager_tab_dialogs.jinja2#L35 into a "row-fluid"

I would also recommend moving the "<input type="hidden" data-bind="value: spoolDialog.spoolItemForEditing.version" value="">" that is added up below the form and not inside the row - ie ending up with something like this:
`<form class="form-horizontal" id="spool-form">
<input type="hidden" data-bind="value: spoolDialog.spoolItemForEditing.version" value="">
<div class="row-fluid">...`

This fixes the layout issues and also make the html more "clean" - because of the way bootstrap uses :before and :after alot.

lightmaster, 2021-01-27:
@LazeMSS Appologies, I didn't necessarily mean that it was UI Customizer that was messing it up, just that with both extensions installed it become messed up for some reason that's way over my head.

LazeMSS, 2021-01-27:
No to apologise 😄

Btw I have no idea why the pasted code doesn't show

VibroAxe, 2021-01-27:
@LazeMSS i'm curious, if it's not UI customiser why does it work fine when UIC isn't installed/enabled?

LazeMSS, 2021-01-27:
Well it is UI Customizer - but it basically just includes the bootstrap responsive and that CSS file contains futher specs for mobile devives etc. - As far I could tell then having maxed out the span width (the maxium is 12) inside a row that is then contained inside a modal causes layout flow problems - furthermore a hidden input is introduce that hits one of the ":before".

Or thats what it looks like when I did a quick install - I didn't do a full test of everything.

But the old bootstrap octoprint is running has alot of quirks. The reason I think it should be fixed inside Spool manager is to make that HTML be more correct and thereby in general make the html more "compliant" - I could include a specifik "hack" inside UI Customizer but that wouldn't solve the real cause of the problem, ie the "bad" html

LazeMSS, 2021-01-27:
Futhermore if I have the time in one of the comming weeks I will be installing the plugin and I might send some UI/UX PR :)

moong8te, 2021-01-28:
Displaying issue with SpoolManager 1.3.3
Issue did not exist in 1.2.0

I also see the same issue.

I am using 2 seperate Pi's with two seperate Octoprint installations.
Both are running the same configuration and hat Spool Manager 1.2.0 installed on them not showing the problem.

I just updated one instance to Spool Manager 1.3.3 and can see the same cutoff of the spool info window.

My findings so far:

If I zoom down in chrome browser all the way down to 50% the info start to be displayed again although far too small to be read. At 33% all info is displayed but then unusable.

I added screenshots from the same Notebook = resolution and browser are the same.
I then opened spool manager 1.2.0 on my first pi and the updated spool manager 1.3.3 on my second pi in direct comparison.

**Screenshot 1.2.0**
[image]

**Screenshot 1.3.3** (see the hard cut and darker area after the barcode)
[image]

**Themeify (spoiler - I later disabled UI customizations and the issue remains)**
I also added screenshots of themeify that I use and my settings.
My settings are stretching the UI over the screen which is intended.
This leads to the spool info display much larger and in one scrollable window instead of two clolumns. Also this is intended.

**Screenshot themeify settings**
[image]

In 1.2.0 and before this worked just fine.

With the current 1.3.3 however there is this cutoff right around the barcode symbol.
Dislabling the barcode in the settings makes no difference.

VibroAxe, 2021-01-28:
> Well it is UI Customizer - but it basically just includes the bootstrap responsive and that CSS file contains futher specs for mobile devives etc. - As far I could tell then having maxed out the [...]

That makes much more sense. I'm wondering if it's actually the input which is breaking it and moving the hidden input into the span6 would resolve the issues? Although this was added in 1.2

moong8te, 2021-01-28:
I see you are currently circling in on UI customizer as part of this issue.
But in my case the issue does not exist on my second pi with seperate octoprint installation and Spool Manager 1.2 (even I have the same UI customizations running).

And the issue also persists after I have deactivated UI customizations on the system I have already upgraded to 1.3.3.

I wonder how could this still be an issue of UI customization then?

As I have two systems running with UI customizations and due to one still on 1.2 and one already on 1.3.3 and showing the issue - I am happy to provide more info if you point me into the right direction.

LazeMSS, 2021-01-28:
> That makes much more sense. I'm wondering if it's actually the input which is breaking it and moving the hidden input into the span6 would resolve the issues? Although this was added in 1.2

I did not spend to much time on testing it - it could be just that.

LazeMSS, 2021-01-28:
> I see you are currently circling in on UI customizer as part of this issue.
> But in my case the issue does not exist on my second pi with seperate octoprint installation and Spool Manager 1.2 (even I [...]

You can try from your browser developer console (F12 on windows) and then run:
`$('input[data-bind="value: spoolDialog.spoolItemForEditing.version"]').remove();`

moong8te, 2021-01-28:
Funny thing is just opening the developer console resizes the main and the missing info below the barcode is revealed again.

**no console**
[image]

**with console**
[image]

After closing the console the issue is back again. But it is a nice workaround or useful for testing the issue very simply.

I also tried your command:
$('input[data-bind="value: spoolDialog.spoolItemForEditing.version"]').remove();

It returned:
S.fn.init [prevObject: S.fn.init(1)]

But unfortunately it does not change the situation in any way.

LazeMSS, 2021-01-28:
Okay - my octoprint is turned off right now - I will try and test later if I have the time.

lightmaster, 2021-01-28:
@moong8te when changing settings that you expect will change the HTML/CSS of the site, make sure you're pressing `Ctrl+F5` to force a full refresh including wiping the browsers cache data for the site. Could be that when you disable UI Customization, it's not completely clearing the CSS that it modifies.

moong8te, 2021-01-28:
> @moong8te when changing settings that you expect will change the HTML/CSS of the site, make sure you're pressing `Ctrl+F5` to force a full refresh including wiping the browsers cache data for the [...]

Thank you for pointing that out. I did try shift + F5 before which should also ignore the cache content.
So without customizations and with CTRL + F5 it looks a bit different but still not right.

[image]

Part of the QR is missing and most of the comment field is missing.
If I then zoom down to 80% I get this:
[image]

But as you can guess text is rather small and I much more like the customized UI as it greatly enlarges the image size of the webcam.

Looking forward to a fixed version of Spool Manager as the current additions look great!

OllisGit, 2021-02-02:
btw. you can deactivate the QR-Thumb in the settings, then the page is a little bit smaller ;-)

moong8te, 2021-02-03:
> btw. you can deactivate the QR-Thumb in the settings, then the page is a little bit smaller ;-)

Hello Olli,

I did test that as part of my trouble shooting.
But then the page is just missing the QR code and still it looks like in my former post like below.

In 1.3.3 the lower darker shaded area in my case takes up a very large amount of space leading to this situation.
This was not the case with 1.2.0.

[image]

jbbandos, 2021-02-04:
Hi, I don't have UI Customizer, and I have this same issue on two separate octoprint instances on two different raspberry PI (one is a 3, the other a 4).
At 67% zoom I can see the whole dialog, but with the text box too small and moved left:
[image]

At 100% it is as documented by @moong8te:
[image]

moong8te, 2021-02-04:
Hello jbbandos,

thank you for your addition.

I think UI customizations and the newly added QR code came into play as first guesses in relation to this problem.

In the meantime I believe it is clear that this is not UI customization related.
I did als well remove all UI customization during my trouble shooting and the issue was still showing.

It also is not QR code related as disabling this function also does not resolve the issue.

The actual fields containing the data is there according to my analysis.
For example you can zoom out (to in your case 67%) and mark the number for used lengh.
When you then zoom in to 100% again you can see the numbers shining through.

It seems like that the background layer is not in the background and covering the information that is actually there and as such blocking them for the user to see.

But then again it is Olli who is the expert on this.

OllisGit, 2021-04-15:
duplicated, see #142

closed 2021-04-15 by OllisGit

lightmaster, 2021-04-15:
> duplicated, see #142

Did you mean to mark the other newer post as a duplicate of this one and close that duplicate one, rather than close the original post that's has a lot of discussion about this issue?