OllisGit/OctoPrint-SpoolManager #114: QR Code Scanning not Working

VibroAxe, 2021-01-26:
By changing the selection implementation of /QR/<id> the QR code is reliant on the ui picking up the #tab_plugin_SpoolManager-spoolId<ID> feature. This does not currently seem to be working and browsing the the QR code just ends up back at the home page

At first I thought this was an issue with a conflicting plugin but I've disabled all my plugins and still have the same issue. Looking through the git commit I can see the tabChange code which should be processesing the -spoolID tag but this doesn't seem to be firing on page load within an "invalid" (#tab_plugin_SpoolManager-spoolId21) tab code because the tabChange event never fires

VibroAxe, 2021-01-26:
I think looking at this the issue comes from the following line

```
if ("#tab_plugin_SpoolManager" == current){
```

When the page loads for the first time current is not guaranteed to be `#tab_pluginSpoolManager` so the spool selector logic doesn't work. (In fact it's highly unlikely to be SpoolManager unless we have set SpoolManager to be the first tab)

However you are already checking by href tag later in the function, and as soon as we move tabs the href updates automatically, so I think this extra if is unnecessary

VibroAxe, 2021-01-26:
@OllisGit I've got a fix for this incoming in a PR, however, some food for thought

Now the QR code is taking you to the edit screen, is it best to auto select the spool, or would the "select for printing" button in the spool editor be a better way of "activating" the spool for print, so the QR just brings up the inventory, or are we best leaving it as it is with auto selection on scan?

VibroAxe, 2021-01-26:
@OllisGit POC for a resolution for the above, 

testing version is live on https://github.com/VibroAxe/OctoPrint-SpoolManager/archive/master.zip for debug

OllisGit, 2021-01-27:
Hi @VibroAxe,

thanks again for the PR. My fix looks similar and I was surprised about the tab-handling, but after I moved the tab to a different position I could reproduce the issue.

This PR will be in the next release, maybe without the direct activation of the spool in the sidebar:
https://github.com/OllisGit/OctoPrint-SpoolManager/pull/115/files#diff-3ee85462fec5e5ebdcd8f1a9664b487087c9f5925cc5f77cd04b4f0ae90aceddR744

Regarding the behaviour after the QR-Code is scanned, I want to tryout the GitHub-Discussion feature, but  I need some days to understand the features and how to structure future plugin discussions.

VibroAxe, 2021-01-27:
Brilliant, thanks

Out of interest, why wouldn't you activate the sidebar? Otherwise the sidebar is out of sync with the actual configuration?

Github discussion seems like a sensible usecase, I wait to see the question

closed 2021-01-27 by VibroAxe