OllisGit/OctoPrint-SpoolManager #116: Problems importing .csv file into database

5av10ur, 2021-01-27:
I'm trying to import a database of spools I have created and have had some issues with the importing process. 

I downloaded the sample .csv file and added my records to it ready to import. I selected to replace the existing database rather than append as I was starting from new. When I first started to import it threw an error: "Error parsing value '02.06.2019' for field 'Purchased on [dd.mm.yyyy]': invalid literal for float() 02 06 2019" despite the field being formatted as dd.mm.yyyy as per the other date fields.

After a bit of investigation it turns out that, despite the field header stating that the format for this column is [dd.mm.yyyy] it actually should be [dd.mm.yyyy hh:mm] so I added 00:00 to each field entry which cleared the parsing error. Now, however, it just hangs when parsing the final line. The pop-up window shows "Status: running" and "Parsing/Importing line: 11" (which is the total number of lines in the .csv file including header) but never moves on from this. 

To solve this I had to select "Append to Database" rather than Replace. I have no idea why Replace didn't work but I did get the plugin to import the database eventually. I post this in case others have a similar problem and this helps them.

OllisGit, 2021-02-02:
Hi @5av10ur ,

thx for reporting the issue!
Could you please attach the csv file to the issue, so that I can reproduce the error?

jbbandos, 2021-02-05:
@OllisGit I am getting the exact same issue that @5av10ur has, that replacing the database fails with the popup showing the "Parsing/Importing line:" with the total number of lines.
I am attaching my CSV in case it helps.
[SpoolManager-merged.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/5931544/SpoolManager-merged.zip)

5av10ur, 2021-02-06:
Attached is the .csv file. Let me know if I did anything wrong with this or whether it helps.

OllisGit, 2021-05-24:
Hi @5av10ur!

fyi: Answering via email and attaching a file doesn't  work in GitHub.
You need to use the WebUI to upload a file.

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 the issue should be fixed.

Please test and give me a feedback.

Thx, in advance
Olli