OllisGit/OctoPrint-SpoolManager #117: Dual Head Spool Tracker

ndekker1, 2021-01-28:
Love this tool but I am wondering if it at all possible to assign individual spools to different print heads. I run a Sigmax R19 with the individual print heads. Sadly, unless I am blind which isn't impossible, I cannot see a section to assign a spool to each tool making it hard if I want to print a mirrored or copied object. This isn't an issue as much as a possible helpful feature in the future. Keep up the good work and I really enjoy it!

ecarpenter2, 2021-02-01:
I am wondering the same thing?  I have a Creality CR-X dual extruder and can only seem to make one spool active at a time.

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 the PR from @codingcatgirl is now included. It looks really nice. Thanks again!!!

Please test and give me a feedback.
Olli

ecarpenter2, 2021-05-24:
Outstanding!  Thanks for letting me know.  I will check it out immediately.

Everett E Carpenter PhD MBA
804-723-1005

ecarpenter2, 2021-05-25:
Just FYI,
When I ran a print with a new spool of filament, Spool manager says I used
75g of material, while the actual weight of the print was only 63g.  Not
sure why there would be a difference or if there is a way to correct it
other than weighing the spools after each print.

ecarpenter2, 2021-05-25:
never mind.  I think I just had the density off.

codingcatgirl, 2021-05-25:
Hah, the same happened to me the other day. If you google "PETG density", you get a wrong result. And i was wondering why i had so much filament left after "using up" each spool.

ecarpenter2, 2021-05-25:
What do you use as your densities for PLA, ABS, and PETG?

codingcatgirl, 2021-05-25:
I use the ones from the site linked in the edit tab of spoolmanager.

ecarpenter2, 2021-05-25:
Thanks

Everett E Carpenter PhD MBA
804-723-1005

closed 2021-06-05 by OllisGit