OllisGit/OctoPrint-SpoolManager #119: Weight calculations don't take into account spool weight

Sembazuru, 2021-01-31:
Here is a screenshot in v1.3.3 with a made up spool:
[image]
The imagined spool here was sold as a 1K spool that actually had 1014g of filament on the spool (not an entirely rare circumstance). The empty spool weighs 150g. So, after weighing the unused spool on a scale the user enters that total weight of 1164g into the "Total weight" field. The error shows up here, but for the example I also entered an imagined usage of 1kg, leaving 14g on the spool. But, the calculations for remaining are wrong. After using 1000.0g of filament the remaining should be 14g, not 164g.

Or am I misunderstanding the intended value for the "Total weight" field as gross weight of the brand new spool instead of net weight of the brand new spool?

ndekker1, 2021-02-01:
Yes you are misunderstanding, The total weight is that of the filament, not the filament and spool. I do agree and think it should be just labeled as filament weight but its an easy fix once you understand what it means. Hope this helps!

Sembazuru, 2021-02-01:
If "Total Weight" is just the original Net Weight of the filament, what is the point of keeping track of the spool weight?

ndekker1, 2021-02-01:
Well, if you just weigh your filament/spool all as one then you'll get the problem you have now where the software thinks you have an extra 150g of filament. This can mean that you think you have enough filament left in the roll to print a 70g part when in reality you only have enough filament to print ~1/4 of that part. You can just put the 1000g that the roll says it has but as you have found there is a percentage of error that could be seen which could be both positive or negative. So, in the end, it is just to keep a better eye on the total amount of filament.

Sembazuru, 2021-02-02:
But I can't weigh my filament separated from the spool, unless I want to pull all the filament off the spool and put it back on when I'm done. Not exactly practical... :-/ (Remember, 1kg of 1.75mm PLA filament is over 1000ft (over 300m)) Maybe relabel this as "Net Weight" instead of "Total Weight"...

I wonder if a "manual update" button would work. This button would pop up another window with two text entry boxes. One would be a current total weight box to enter in the weight from a scale of the current filament on the spool. This could be pre-populated from the database `remaining` added to the database `spool weight`. This allows the user to compare that number to what they see on the scale and either update it or think to themselves "Meh, that's close enough". The other text entry box would be the spool weight that is pre-populated from the database `spool weight`. This allows updating the spool weight if they get a better number here. (Either an update from the mfg, or an average of multiple spools that the user has already emptied,.) Maybe have on this update window a checkbox that is normally not selected labeled something like "initial entry". If this box is selected then the Total/Net Weight gets updated. If the box is not selected then the Remaining gets updated. (Spool weight is updated regardless, even if it means updating with the same value you don't need any logic to decide if to update it or not.)

(I don't know the GUI engine well enough to know if yet another pop-up window on top of the update/enter spool window would work.)

stevegt, 2021-02-12:
Just ran into this myself -- the English word "Total" means "everything" in any other context -- here this would mean "filament plus spool", so that field is definitely mislabeled.  I had to look at the code to prove to myself that the field label should actually be "Filament Weight" not "Total Weight".  

I would do this:

1. Change the field label to "Filament Weight" to get rid of user confusion.  I'd do this soon, because this is going to be harder to change later if users get used to thinking that "filament weight == total weight".
2. Later add an optional "Total weight" field and have the code subtract spool weight from total to get filament weight.

lightmaster, 2021-02-20:
My 2 cents on how this *should* be implemented:

- Total weight = measured weight of filament + spool before any was used for printing
- Spool weight = estimated weight of spool, calculated by weighting spool with all the original filament on it, so 1kg roll of filament actually weighs 1275g, so you know the spool should weigh 275g
- Remaining Total Weight = weight of filament + spool as measured by a scale, editable so you can easily update your filaments when you weigh them
- Remaining Filament Weight = weight of just the filament that's still available for printing
- Used Filament Weight = weight of just the filament that already been printed (ie, current millage of that spool)

abc2006, 2021-02-22:
And i would appreciate a Button where i am able to enter the total Weigh (e.g. if i remove some Filament manually or ... ) so i am able to weigh my Spool completely and have SpoolManager calculate and modify the actual Values.

melyux, 2021-06-19:
This keeps tripping me up too. The program should do the calculation for you. Enter the actual total weight, spool + filament, and the empty spool weight, and it should subtract it for you and use it as the filament weight.

Noggin01, 2021-07-06:
I just learned this one the hard way myself with a 20 hour print failed at the 18 hour mark. At the very least, the input page should be explicitly clear as to what it is asking for. Now that I've corrected the spool, it shows it at around -5% remaining.

At the very least, the data entry page should be _obvious_ as to the data it is requesting. The entire point of this plugin is to track the filament usage, and I'm quite happy with the _performance_ of the plugin. 

If nothing else, a sanity check could be added using the total length. Prusament gives me the length of the spool. I entered that as 344000 mm. Given a diameter of 1.75mm, I can calculate the cross sectional area as pi * (1.75/2)^2 = 2.4041 mm^2. Multiplying by the length, 344000 mm gives me 826997.5 mm^3, or 826.9975 cm^3, for the volume of filament. Multiplying by the density of 1.27 g/cm^3 yields 1050 grams.

https://prusament.com/spool/?spoolId=e5e3e5b183  

Prusament reports that this spool had 1046 grams on it, for a difference of 4 grams, or 0.38%, from the calculated percentage. Given that I had entered 1237g as the total weight (because that is what the total weight actually was) yields a difference of 187g from the calculated mass, or an apparent error 17.81%.

1. User interfaces should be clear so idiots like me can still get it right
2. If the UI isn't clear, the information should at least be visible so that people that read can get it right
3. If the necessary information isn't visible, then the UI should use whatever information it is given it to try and protect the user from his own stupidity

abc2006, 2021-07-06:
actually, your print should not fail because of missing filament.. my Prusa pauses the print, and as i re-supply with another one, the Print is finished. But the basic problem is the same for me!

cheers, 
stephan

OllisGit, 2022-01-12:
Hi ! 
The mentioned PR is now in the latest release!

Because I have a lot of issues related the weight calculation I close this issue to get a more "up to date overview" in the issue-tracker.
If you found an issue in the latest version feel free to raise a new issue!

closed 2022-01-12 by OllisGit