OllisGit/OctoPrint-SpoolManager #123: Cannot edit spools that have run out (>100%)

moarkaboom, 2021-02-09:
I had a spool end during a print, but Spool Manager obviously had no way to understand, that I changed the spool for a new one during the print. The result is that my ended spool is now shown at 100.9%. I cant edit any material info in that spool now, only option is to set it to inactive.
[image]
[image]

thirschbuechler, 2021-04-13:
if you can't see all the controls, maybe your user is not an admin

[image]

thirschbuechler, 2021-04-13:
you can set used weight to whatever you want

[image]

closed 2021-05-24 by OllisGit