OllisGit/OctoPrint-SpoolManager #126: FEATURE: shared database for two or more OctoPrint systems

ingo916, 2021-02-17:
If possible it would be nice to have a shared database that two or more OctoPrint servers can use. This way when i use a spool on one printer and then another the data is updated and accurate.

The-EG, 2021-03-01:
I have 2 instances of OctoPrint running on the same RPi. 

I've been able to have Spool Manager use the same database for both instances by moving the sqlite database to a common location:
```terminal
$ mv ~/.octoprint/data/SpoolManager/spoolmanager.db ~/
```

 and then symlinking both instances to that one (--basedir for octoprint instances are ~/.octoprint and ~/.octoprint2):

```terminal
$ cd ~/.octoprint/data/SpoolManager
$ ln -s ~/spoolmanager.db .
$ cd ~/.octoprint2/data/SpoolManager
$ ln -s ~/spoolmanager.db .
```

There may be concurrency concerns, but so far it appears to function as desired. Updates from one instance appear in the other, and most importantly, spools appear in both, etc.

If it were possible to specify a full path to the sqlite db in the plugin settings, this type of setup could be easily achieved.

ingo916, 2021-03-01:
that is great! however I am running two seperate RPi's and would like to accomplish the same thing. Maybe like creating a shared folder on my Synology NAS and having both RPi's access the DB from there? Or a shared folder on one RPi?

adamboutcher, 2021-03-04:
> that is great! however I am running two seperate RPi's and would like to accomplish the same thing. Maybe like creating a shared folder on my Synology NAS and having both RPi's access the DB from there? Or a shared folder on one RPi?

Nfs and Sqlite don't always play nice, I can't see it being that hard to rewrite it to support mysql or postgres?

adamboutcher, 2021-03-04:
There looks to be basic support for postgres and mysql already in Database Manager.py you could try tweaking that?

ingo916, 2021-03-05:
I wouldn't know how to do that, lol. I'm not good at coding.

ingo916, 2021-03-06:
I managed to create a NFS share named SpoolManagerDB on my synology server, then on both my RPi's i ran
sudo apt-get install nfs-common
sudo apt-get install portmap

then I created a local mount director on the pi's
sudo mkdir /mnt/SpoolManagerDB
sudo chmod -R 777 /mnt/SpoolManagerDB
sudo chown pi:pi /mnt/SpoolManagerDB

Then i mounted the share by running the following on both RPi's
sudo mount -t nfs 10.8.0.6:/volume1/SpoolManagerDB /mnt/SpoolManagerDB

Then i copied the spoolmanager.db to this mount location. Both RPi's see the file.

How and where do i change the SpoolManager plugin to point to this location for the database?

ingo916, 2021-03-06:
this is the article i used that helped
https://behind-the-scenes.net/mounting-a-shared-nas-folder-onto-a-raspberry-pi/

adamboutcher, 2021-03-08:
NFS and SQLite are known to have issues due to NFS locking; you might find that it doesn't always update correctly.

ingo916, 2021-03-08:
thanks for the info. I guess I will wait for the developer to add the support for a shared DB location.

andrew-newell, 2021-04-07:
It looks like this is a duplicate of #5

Xiondewah, 2021-05-26:
This would be a very useful feature and I think it would encourage donations since the only alternative seems to be either cloud-based or subscription services.

ldiegos, 2021-07-27:
Could be great if you reuse the postgreSQL that your filament manager(inherit from the other project) use. This will help so much to the community that use the old filament manager and have a lot of stuff (very helpfull the import/export tool), but we already have been use to the postgreSQL.

kelvindavis, 2021-10-09:
I would love to be able to have a database for both Pi to pull from.  would make things much nicer.  Big upvote for this

closed 2022-01-25 by github-actions[bot]