OllisGit/OctoPrint-SpoolManager #127: api call from other plugin in python

giloser, 2021-02-18:
Hi,

I develop on the telegram plugin and we had the filament manger manage through telegram with some call to the url.
Could you direct me how I could call your plugin as well?
a list of the spool, a way to select one?
stuff like that?

thanks a lot

OllisGit, 2021-04-17:
Hi @giloser,

well you are not the only one you requested this, see #145

You can use the current UI-Backend-APIs like this one:
http://localhost:5000/plugin/SpoolManager/loadSpoolsByQuery?filterName=hideInactiveSpools&from=0&to=3000&sortColumn=lastUse&sortOrder=desc
And this is definitely not a REST-API !!! It is just an interface between my plugin UI and the backend, nothing more.

You can use the current endpoints, but "at your own risk" ;-) It is not stable, if I need to change it, i will do that (without a release note).

It is planed to implement a public REST-API (or other API-Styles) for other plugin developers, but currently it is far away.
See long term column in backlog: https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1#card-27695833

I need to do a lot of other things in front of that (e.g. support of external database and migrate to a three-tier-layer-architecture, ....)

Sorry, but I am closing this issue, because "it works as expected" and the feature is already on the roadmap and i want to cleanup the issue tracker.

closed 2021-04-17 by OllisGit

giloser, 2021-04-17:
Thanks for the answer I will wait then :-D

OllisGit, 2021-04-19:
Hi @giloser,

I created a wiki-page to start a discussion how the API could look like:
https://github.com/OllisGit/OctoPrint-SpoolManager/wiki/REST-API-Design

Currently it is empty, but maybe you can type your requirements to it.