OllisGit/OctoPrint-SpoolManager #128: Can't set Length Remaining with Decimals

lightmaster, 2021-02-20:
If you set the length remaining using decimals, ie 119,967.7mm, then it will not be saved when you hit the save button. Anything else you have edited will be saved, but length remaining/used will be blank for both. Admittedly keeping the length as a decimal mm is basically useless, but even still, it shouldn't cause it to be saved as blank.

lightmaster, 2021-02-20:
Also, if you enter a number with a comma, it immediately considers that number to be 0, rather than parsing out the comma for the needed calculations. Displaying the numbers with commas would be really helpful, especially when you have 300000mm as a number.

thirschbuechler, 2021-04-13:
bug confirmed. also, it seems the length is not synced with remaining mass, even though the density is provided

closed 2022-01-25 by github-actions[bot]