OllisGit/OctoPrint-SpoolManager #130: [Feature Request] Support filament changes?

abrenoch, 2021-02-21:
It would be really nice if we had the opportunity to swap out the filament being used if an M600 event occurs. One would need the 'printer notifications' working it seems - this would give users the chance to select the new filament when octoprint prompts the user. Though this would all come down to how exactly the weight of the job is being calculated - obviously you would need to be able to determine how much filament was used up to and then after the M600.

Not sure how realistic that is, but wanted to toss the idea out there!

abc2006, 2021-02-22:
+1 for me :)

MrDragon420, 2021-02-25:
FOR SURE

codingcatgirl, 2021-05-15:
Hey, as part of my pull request for Multi-Material support i've added the option to change the filament mid-print: #161 Please check out of that works for you :)

Poporacer, 2021-05-19:
@codingcatgirl How can I load your version of the plugin to test?

OllisGit, 2021-05-19:
@Poporacer go to the plugin-manager and enter this url:
https://github.com/codingcatgirl/OctoPrint-SpoolManager/archive/refs/heads/multi-material.zip

I am currently reviewing the PR. It looks really really good, so I think it will be merge in the next days.
It will take some time to review, because @codingcatgirl  and myself did a lot of changes in the sourcecode.

PR #161 (2021-05-19): Multi-Tool/Multi-Material support
Hi! 

Since i really needed Multi-Tool support in this Plugin right now, i decided to just implement it myself :)

Simply put: It's done and this is how it looks: 

[image]

List of Changes/Features:
- Revamped entire backend to work with spools for multiple tools (fixes #46, #153)
- Allows manual filament-change mid-print. If a new spool gets selected mid-print, the user gets asked whether they would like to commit the filament usage of the print so far to the previously selected spool.  (fixes #130)
- Ensures that a spool can't be selected in multiple tools
 [... cut]
closed 2021-05-19 by OllisGit