OllisGit/OctoPrint-SpoolManager #132: Can't unselect spool

Lerxst151, 2021-02-25:
I have quick print that I want to do on a spool that I have not put in Spool Manager.
I no longer can unselect the current spool. Clicking on the check box next to the spool does not unselect it, it just takes me to the edit page of that spool.

Lerxst151, 2021-03-11:
anyone????

OllisGit, 2021-04-18:
Hi @Lerxst151,

can you see the X on the right side. Click on this and the spool should be unselected. 
[image]

Lerxst151, 2021-04-19:
No the X does not show up anymore in Chrome or FireFox

Lerxst151, 2021-05-21:
I figured out the issue... If you put too many characters in the Displayname field then it runs to the end of the box and you cannot see the X.

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 I did some UI-Improvements.

Please test and give me a feedback.

Thx, in advance
Olli