OllisGit/OctoPrint-SpoolManager #133: [Feature Request] Multi Spool Support

MadeforMaking, 2021-03-01:
Regarding machines that have either multiple tools or multiple filaments loaded.

NOT multiple material prints as I would imagine that there would be a lot more work involved with this which I'd say should be a separate issue. This request is mainly regarding the single-mode on multi-material printers. I'm speaking from experience with the Prusa I3 MMU2. 

SpoolManager does its job at selecting spools and is not causing problems, however, I suggest a mode (option) dedicated to multi-material machines, where multiple spools can be displayed and "loaded" with a single option checkbox to "select" the spool to load. My current workflow includes [ tkoecker/OctoPrint-Mmu2filamentselect ](https://github.com/tkoecker/OctoPrint-Mmu2filamentselect) where I get prompted to select what spool to load.

[image]

Here is an example image I quickly put together. The left checkboxes decide what filament will be loaded and you can change the filament on the right. This is only for example and obviously, you can't duplicate your filament 😮!

To work across platforms the number of loading spots could be customizable for the user given different printer setups.

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 the PR from @codingcatgirl is now included. It looks really nice. Thanks again!!!

Please test and give me a feedback.
Olli

MadeforMaking, 2021-06-10:
Testing now. 

Would be great if SpoolManager could automatically select filament for "single-tool" prints. I currently use the [OctoPrint-Mmu2filamentselect](https://github.com/derpicknicker1/OctoPrint-Mmu2filamentselect) plugin to select the filament, without having to manually select on the printer. Maybe @derpicknicker1 could share some insight, if this could be done?

Again, all my experience is with the Prusa i3 MMU2 and other multi-tool printers would need to be considered.

Otherwise great, I have run a "single-tool" print here and it seemed to function correctly with v1.4.3. 👍 

Still to test a "multi-tool" print.