OllisGit/OctoPrint-SpoolManager #134: [Feature Request] Multi Material Support

MadeforMaking, 2021-03-01:
See #133 first, I'd say this request would be hard to implement. But let me know.

Similar idea as #133 but with the support of a multilateral print. If an MM (Multi-Material) print is detected then the checkboxes may change to dropdown menus stating what "Tool" each spool represents in the print.

[image]
Here's an image of PrusaSlicer and how it handles the materials or spools. 

Now it handles the MM aspect of printing and no major problems with physically printing. There is no way however for octoprint to handle the amount of filament used, this is where this plugin I hope can come in. If I have all the materials loaded into the machine but not in the correct "Tool" or slot or order, I would like to be able to change this using this plugin. This feature might also be work for layer changes in prints. 

Hopefully, that explains my thoughts, let me know yours and if you need more information, even if it's able to be implemented.

WoLpH, 2021-03-26:
Multiple material support would be great. It's my main reason for still using filament manager

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 the PR from @codingcatgirl is now included. It looks really nice. Thanks again!!!

Please test and give me a feedback.
Olli

MadeforMaking, 2021-05-25:
I will, Not in the next couple of days, but will when I get back to printing. Thanks

skarfacegc, 2021-07-02:
This isn't working for me.  Running version 1.4.3 (if you're looking for test feedback)

codingcatgirl, 2021-07-02:
"doesn't work" is never a good error description. in order for your
feedback to actually be useful, i recommend actually describing the issue
you are experiencing. :)

skarfacegc, 2021-07-02:
Sorry ...

skarfacegc, 2021-07-02:
Sorry ... version 1.3, palette2 with up to date firmware

skarfacegc, 2021-07-02:
Apologies ... wasn't thinking of detail.  :). Opened a more detailed description in  #187 (will put here as well)
Version 1.4.3, palette2 with up to date firmware.  Octoprint up to date (as is spool manager / print job history), successful prints via octoprint and canvas, haven't tried p2pp yet.

Filament assigned to all 4 extruders in octoprint
Job is recorded in printjobhistory
All of the filament use is assigned to the spool in tool0

Via printjobhistory I do see that the usage appears to be getting reported to octoprint (assuming that is what  slicer settings under tool management is reporting, sample below)

```
 ; DisplayLayerProgress_layerIndicatorProcessed = true
;    Ext 4 = 8589.94 mm
;    Ext 3 = 8708.16 mm
;    Ext 2 = 5271.16 mm
;    Ext 1 = 7249.56 mm
; END_LAYER_OBJECT z=15.000
; Dwell time remaining = -111.647316 s
; time estimate: pre = 22.328001 s, post = 116.647316 s
; path_width=0.4000
```

Will happily post any other relevant detail

skarfacegc, 2021-07-09:
These seem to be the relevant log lines.  Just turned on DEBUG level logging, will followup with more if anything new/relevant shows up.  Any pointers on log lines to look for?  Already stripped out the database lines.

```
021-07-08 15:09:57,886 - octoprint.plugins.SpoolManager - INFO - Tool 0: Extruded filament length: 966.0436400000027
2021-07-08 15:09:57,886 - octoprint.plugins.SpoolManager - INFO - Tool 0: Current Spool filament length: 234583
2021-07-08 15:09:57,886 - octoprint.plugins.SpoolManager - INFO - Tool 0: New Spool filament length: 235549.04364
2021-07-08 15:09:57,935 - octoprint.plugins.SpoolManager - INFO - Tool 1: No filament extruded
2021-07-08 15:09:57,936 - octoprint.plugins.SpoolManager - INFO - Tool 2: No filament extruded
2021-07-08 15:09:57,936 - octoprint.plugins.SpoolManager - WARNING - Tool 3: No spool selected, could not update values after print
2021-07-08 15:28:58,767 - octoprint.plugins.SpoolManager - INFO - Tool 0: Extruded filament length: 237.23651000000112
2021-07-08 15:28:58,767 - octoprint.plugins.SpoolManager - INFO - Tool 0: Current Spool filament length: 235549
2021-07-08 15:28:58,767 - octoprint.plugins.SpoolManager - INFO - Tool 0: New Spool filament length: 235786.23651
2021-07-08 15:28:58,843 - octoprint.plugins.SpoolManager - INFO - Tool 1: No filament extruded
2021-07-08 15:28:58,843 - octoprint.plugins.SpoolManager - INFO - Tool 2: No filament extruded
2021-07-08 15:28:58,844 - octoprint.plugins.SpoolManager - WARNING - Tool 3: No spool selected, could not update values after print
2021-07-08 15:49:27,111 - octoprint.plugins.SpoolManager - INFO - Tool 0: Extruded filament length: 475.464440000004
2021-07-08 15:49:27,112 - octoprint.plugins.SpoolManager - INFO - Tool 0: Current Spool filament length: 235786
2021-07-08 15:49:27,112 - octoprint.plugins.SpoolManager - INFO - Tool 0: New Spool filament length: 236261.46444
2021-07-08 15:49:27,156 - octoprint.plugins.SpoolManager - INFO - Tool 1: No filament extruded
2021-07-08 15:49:27,156 - octoprint.plugins.SpoolManager - INFO - Tool 2: No filament extruded
2021-07-08 15:49:27,156 - octoprint.plugins.SpoolManager - WARNING - Tool 3: No spool selected, could not update values after print
2021-07-08 21:34:28,479 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.SpoolManager level to 10
2021-07-08 21:34:42,182 - octoprint.plugins.SpoolManager - DEBUG - API Load all spool
2021-07-08 21:37:25,026 - octoprint.plugins.SpoolManager - DEBUG - API Load all spool
```

codingcatgirl, 2021-07-09:
If everything lands on the first spool, that's probably a problem with the new odometer that was added after my pull request =(

skarfacegc, 2021-07-09:
not frustrated (hard to get angry at free software most of the time) just want to help fix.  Any other relevant logs?  I'm also using the palette via canvas (not sure if that matters here at all or not)

codingcatgirl, 2021-07-09:
Oooh, the canvas (i had to look it up but i have seen it before). Well… does it make the tool change commands look different in Gcode? (compared to simple T0, T1, T2…) That might be the problem then.

codingcatgirl, 2021-07-09:
That, or something about the way it works means that the tool change command gets intercepted and swallowed (not sent to the printer) before the SpoolManager plugin can pick up on it.

skarfacegc, 2021-07-10:
I do see (what I presume to be) correct usage stats in the slicer details from printjob history (above).  So /something/ gets it.  :) Not sure if in a parsable format etc.  I can look for stuff in the gcode if you can point me in the right direction.

skarfacegc, 2021-07-10:
I can also post the gcode somewhere if that's helpful

closed 2022-01-24 by github-actions[bot]