OllisGit/OctoPrint-SpoolManager #139: [Feature Request] Saving Filament Consumption data while printing

KineticEcho, 2021-03-22:
Hi, I really love this plugin!
I was printing a very large file and, sadly, I had a power outage ... many power outages... I have now an UPS attached to the printer and Octopi device but it would be nice if the filament consumption data was saved during the print process. So, in case of a power outage, I could know a rough estimate of the filament remaining (maybe with an alert, to remember the event!)

staffert, 2021-03-26:
That is a good idea, may I also suggest that we decide one way or the other, as to how to depict a used spool?
is it **0% left (as in the sidebar display)** _or is it_ **100% (as  per the data table presentation in the spool manager window)**?
I find it confusing to switch your brain over, when quickly trying to deal with print jobs and material management.
Just a small detail for a really great plugin to make it even better.
Cheers.

OllisGit, 2021-05-24:
Hi @staffert,

in the next release there will be tooltip to make it a little bit clear.
Also, there is an other ticket about customizable the sidebar spool-label, see #83

OllisGit, 2021-06-05:
Hi @KineticEcho,

that is not easy to implement, because I can't detect a poweroutage, also writing each extrusion change to the sd-card is also not a good approach.