OllisGit/OctoPrint-SpoolManager #140: Error when deleting printjob

hlipka, 2021-03-24:
I have seen this recently (I'm on 1.3.3). When doing a print, and deleting the GCode file after its finished, I get this error:
`2021-03-24 19:26:10,094 - octoprint.printer.standard.job - INFO - Print job deselected - user: n/a
2021-03-24 19:26:10,122 - octoprint.plugin - ERROR - Error while calling plugin SpoolManager
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/__init__.py", line 445, in on_event
    self._on_file_selectionChanged(payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/__init__.py", line 383, in _on_file_selectionChanged
    if ("origin" in payload and "path" in payload):
TypeError: argument of type 'NoneType' is not iterable
`
The file is not deleted, and OctoPrint seems to hang afterwards. E.g. I cannot delete other file, or start a print. Opening the settings work, butr not saving them.
 When I restart the RasPi and delete the file, it works again (and I can delete the file).

OllisGit, 2021-04-18:
Hii @hlipka,

thx for reporting.
Issue will be fixed in next release.

BR
Olli

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 the issue should be fixed.

Please test and give me a feedback.

Thx, in advance
Olli