OllisGit/OctoPrint-SpoolManager #141: Can we add a filter or sort option for material type?

chemixt, 2021-03-25:
I would be really nice if I could sort or filter  my spools by material.  

Also an inventory printout would be nice when I take a trip to Microcenter for restocking.

OllisGit, 2021-05-19:
- I was thinking about something like this for filtering
[image]

- Sorting for material is already there
[image]

- How should a print out look a like? Please make a suggestion.

egar0, 2021-05-20:
This probably should be a separate request, but as long as we are talking sorting and filtering...

I'd also like to see the ability to define the default view or to save last view parameters.  For example, I typically want to hide empty spools and sort spools by last use.  Material type could factor into this as well.

chemixt, 2021-05-21:
I like the new filters.  That should help tremendously.  The inventory report should look link this.  It should echo the same sort and filters selected before printing. Or if you can export to excel or .csv then we can sort or filter in excel. 

[image]

chemixt, 2021-05-21:
I also rfeplied on git hub.

I like the new filters. That should help tremendously. The inventory report
should look like this. It should echo the same sort and filters selected
before printing. Or if you can export to excel or .csv then we can sort or
filter in excel.

[image: inventory report]
<https://user-images.githubusercontent.com/65088878/119064569-3bdacb80-b9a1-11eb-8655-8a9be8140a9f.JPG>

chemixt, 2021-11-26:
See new issue "ERROR MESSAGE when printing #218" regarding version 1.5.0 Spool Manager.  This is regarding new issues with errors when printing on Makerbot 2X.  (and maybe other printers). 

https://github.com/OllisGit/OctoPrint-SpoolManager/issues/218

OllisGit, 2022-01-07:
CSV-Export tracked here #209

closed 2022-01-07 by OllisGit