OllisGit/OctoPrint-SpoolManager #142: Filament fields page cutoff

keving98g, 2021-03-26:
I am unable to access some of the fields on the bottom of the screen. This is the only plugin with this behavior and is not effected by an update, different (new and old) filaments. It could be specific to my Octoprint setup but I believe I have eliminated other variables.
[image]
OctoPrint: 1.5.3
SpoolManager Plugin: 1.3.3

bertenvdb, 2021-04-08:
Same here

[image]

thirschbuechler, 2021-04-13:
which browser are you using? maybe some adblocker cuts it off, I'm on the same versions and everything seems fine (see reply on #123)

thirschbuechler, 2021-04-13:
#112

bertenvdb, 2021-04-15:
- Chrome: 89.0.4389.114
- OctoPi: 0.18.0
- OctoPrint: 1.6.0rc1
- SpoolManager: 1.3.3

I have uBlock Origin but disabled. 
Tried in incognito without uBlock.
Created a new Chrome user and tried without plugins.

thirschbuechler, 2021-04-15:
is it maybe caused by zoom level and UI customizer as in #112 ?

OllisGit, 2021-04-18:
Because I can't reproduce the issue I need more infos:

* display resolution
* browser zoom-level
* operating system (e.g. Windows, MacOS)
* Desktop, Tablet or Phone

Rebel154, 2021-04-19:
This problem occurs on my PC with both Firefox (latest) and Edge (latest).

Resolution: 1920x1200

Browser Zoom: 110%. If I reduce the zoom level down in steps to 50%, the missing elements start to progressively reveal themselves. If I increase the zoom level to 200% in steps, again, the missing elements progressively reveal themselves.

OS: Windows 10 Pro build 19042.928 on desktop PC

Plug-ins:

|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Cancel Objects (0.4.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_cancelobject
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Dashboard (1.18.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.26.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.2.4) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  OctoPod Plugin (0.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octopod
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrintTimeGenius Plugin (2.2.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  PrusaSlicer Thumbnails (0.1.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prusaslicerthumbnails
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  SpoolManager Plugin (1.3.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager
|  UI Customizer (0.1.5.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_uicustomizer
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
|  WebcamTab (0.2.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_webcamtab

On my iPad (2020 iPad Air 4), the Spools page loads OK in both Firefox and Safari.

Hope this helps.

melrhombus, 2021-04-20:
I have the exact same issue on Brave
Disable UI Customizer and issue goes away

LazeMSS, 2021-04-23:
@OllisGit its caused by this:
https://github.com/OllisGit/OctoPrint-SpoolManager/blob/0e86e1b32dd06646dabb3e717de7d22458da4942/octoprint_SpoolManager/templates/SpoolManager_tab_dialogs.jinja2#L167

That hidden input breaks the before stuff of the way the responsive bootstrap works. If you move that to right after the form (https://github.com/OllisGit/OctoPrint-SpoolManager/blob/0e86e1b32dd06646dabb3e717de7d22458da4942/octoprint_SpoolManager/templates/SpoolManager_tab_dialogs.jinja2#L167)  it will work and also the code would be more clean.

I would also suggest you change the https://github.com/OllisGit/OctoPrint-SpoolManager/blob/0e86e1b32dd06646dabb3e717de7d22458da4942/octoprint_SpoolManager/templates/SpoolManager_tab_dialogs.jinja2#L165 to a row-fluid to better support mobile/tables.

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 I did some UI-Improvements. @LazeMSS thanks a lot for your suggestions!

Please test and give me a feedback.

Thx, in advance
Olli

Rebel154, 2021-05-24:
@OllisGit

The missing fields problem is back with 1.4.0. All other plugins are the latest versions of my list from above. Zooming out to 67% or in to 200% gradually  reveals the missing fields in a single column.

OllisGit, 2021-05-24:
hmm..then the improvement from @LazeMSS didn't help.
@Rebel154 please make a screenshot...and try to disable other plugins maybe there is a (still) a collision between some plugins.

Rebel154, 2021-05-25:
@OllisGit

Just tried with 1.4.1 as well and the issue persists. I have restarted the Raspberry Pi and Octoprint.

Updated Plugin List

|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Cancel Objects (0.4.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_cancelobject
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Dashboard (1.18.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.26.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Filament Manager (1.7.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentmanager
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.2.4) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  GPIO Control (1.0.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiocontrol
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  OctoPod Plugin (0.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octopod
|  Pi Support Plugin (2021.3.26.post2) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrintTimeGenius Plugin (2.2.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  Prusa MMU2 Select Filament (1.0.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_mmu2filamentselect
|  Slicer Thumbnails (1.0.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prusaslicerthumbnails
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
| !SpoolManager Plugin (1.4.0rc1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager
|  UI Customizer (0.1.5.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_uicustomizer
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
|  WebcamTab (0.2.11) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_webcamtab

My screenshots are the same as the first post above. I disabled all plug-ins except UI-Customiser and the issue persists. Disabling Ui-Customiser clears the issue.

HTH.

diedder, 2021-05-25:
Hi,

in 1.4.2 the fields are underneath the spool window
[image]

LazeMSS, 2021-05-25:
Agreed - something is more broken now

LazeMSS, 2021-05-25:
Not sure why but my right colum has span12 and not span6

LazeMSS, 2021-05-25:
it must be something in my code I think

LazeMSS, 2021-05-25:
@OllisGit I have just released a new version - was a bit rushed but didn't want your plugin to get bad rep due to my "hacks"

OllisGit, 2021-05-25:
Hi @LazeMSS,

I installed the latest version 0.1.5.9 (wow, 4 digits) and can confirm that the spool-item dialog is "almost" fixed (the active checkbox is not in the upper right corner).

But I also tested the PrintJob-Dialog (V1.14.0) and this dialog is still broken. Everything is in one column.
hmmm...to be honest I have no clue whats going on, I am not a UX-Guy.
[image]

achampion, 2021-05-25:
Changing from `span12` to `span6` fixed things on my system.

jbienz, 2021-05-25:
Looks like this may have been fixed with a UI Customizer update posted 4 hours ago:

https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.9

After updating to SpoolManager 1.4.3 and UI Customizer 1.5.9, my UI appears to have returned to normal. Not sure it fixes all of the issues listed above though, so additional testing by those affected would be helpful.

LazeMSS, 2021-05-26:
> Hi @LazeMSS,
>
> I installed the latest version 0.1.5.9 (wow, 4 digits) and can confirm that the spool-item dialog is "almost" fixed (the active checkbox is not in the upper right corner).
>
> But I also [...]

I hopefully have a couple of hours this weekend. I will try and do a rework of the UI and send you a pr

OllisGit, 2021-05-26:
Hi @LazeMSS,

I played around with the chrome dev-tools and saw that your plugin overwrite the standard bootstrap 'span' classes (puhh, is that the right strategy). 

For the PJH-Dialog I disabled the span6 class and now I can see the two columns again.
[image]

As I mentioned I am not a UI-Export, but I think this is not a good way to do. Of course you did an AMAZING job to do a total redesign of the default UI, but are there some other ways to do?

Maybe this is also a question to the OP-Community how to separate Plugin UI-Stuff, is "Shadow Dow" the way to go
https://developer.mozilla.org/en-US/docs/Web/Web_Components/Using_shadow_DOM

I appreciate it very much that you try to find free time to analyse the issue a little bit deeper and (probably) create a PR!

BR
Olli

LazeMSS, 2021-05-26:
My plugin basically just include the standard bootstrap responsive files. It's a standard way of doing it in bootstrap.

Your dialog is basically to crowded for at bootstrap modal dialog.

I will do my best to make a clean solution for you ASAP.

LazeMSS, 2021-05-26:
https://getbootstrap.com/2.3.2/scaffolding.html#responsive

diedder, 2021-05-26:
> @OllisGit I have just released a new version - was a bit rushed but didn't want your plugin to get bad rep due to my "hacks"

fixed at my Safari

LazeMSS, 2021-05-26:
@OllisGit  don't know what PrintJob-Dialog is?

LazeMSS, 2021-05-26:
> > @OllisGit I have just released a new version - was a bit rushed but didn't want your plugin to get bad rep due to my "hacks"
> 
> fixed at my Safari browser on MacBookPro

It also works fine for me :(

OllisGit, 2022-01-07:
No feedback, please raise new issue if needed

closed 2022-01-07 by OllisGit