OllisGit/OctoPrint-SpoolManager #145: ERROR: SPM:DatabaseManager on loadAllSpoolsByQuery API Call

UnchartedBull, 2021-04-13:
Hi,

I'm currently trying to integrate Spool Manager into OctoDash. I do seem to have hit a roadblock though. If I call the `loadAllSpoolsByQuery` endpoint without any body at all I receive just partial response without any spools:

```
{
    "allSpools": [],
    "catalogs": {
        "labels": [
            ""
        ],
        "materials": [
            "",
            "PLA",
            "PLA_plus",
            "ABS",
            "PETG",
            "NYLON",
            "TPU",
            "PC",
            "Wood",
            "Carbon Fiber",
            "PC_ABS",
            "HIPS",
            "PVA",
            "ASA",
            "PP",
            "POM",
            "PMMA",
            "FPE"
        ],
        "vendors": [
            "",
            "mock"
        ]
    },
    "selectedSpool": null,
    "templateSpool": null,
    "totalItemCount": 8
}
```

OctoPrint also shows an Error in the top right-hand corner. I don't seem to be able to find any docs about the API besides the code. From what I understood the method should work fine without any query at all, but I might be wrong here. I've also attached the octoprint.log, don't know if it's helpful or not :). Oh I'm also testing this on a virtual printer running in a docker container, I can create spools and stuff, so I assume that sqlite is working, but I don't know whether this has any other implications.

[octoprint (2).log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/6307059/octoprint.2.log)

I also think it would be great if there would a be a separate endpoint to just query the currently active spool (at least I wasn't able to find one), this should save a couple of database calls. And I think a little bit of documentation might be great since the endpoint naming doesn't seem to be very REST-like, which made it kind of hart to find the correct APIs, but that's just the icing on the cake :)

Thank you!

OllisGit, 2021-04-15:
Hi @UnchartedBull,

well you need to provide the mandatory fields, like this. Then you will receive the data
```
http://localhost:5000/plugin/SpoolManager/loadSpoolsByQuery?filterName=hideInactiveSpools&from=0&to=3000&sortColumn=lastUse&sortOrder=desc
```
And this is definitely not a REST-API  !!! It is just an interface between my plugin UI and the backend, nothing more.

You can use the current endpoints, but "at your own risk" ;-) It is not stable, if I need to change it, i will do that (without a release note).

It is planed to implement a public REST-API (or other API-Styles) for other plugin developers, but currently it is far away.
See long term column in backlog: https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1#card-27695833

I need to do a lot of other things in front of that (e.g. support of external database and migrate to a three-tier-layer-architecture, ....)

Sorry, but I am closing this issue, because "it works as expected" and the feature is already on the roadmap.

closed 2021-04-15 by OllisGit

UnchartedBull, 2021-04-19:
Well not the news I hoped for, but news 😄 Totally understandable though, let me know when there is something available / ready to test, as this is one of the most requested features in OctoDash now :)

OllisGit, 2021-04-19:
Hi @UnchartedBull,

I created a wiki-page to start a discussion how the API could look like:
https://github.com/OllisGit/OctoPrint-SpoolManager/wiki/REST-API-Design

Currently it is empty, but maybe you can type your requirements to it.

UnchartedBull, 2021-06-10:
Hi, sorry it took me sometime to get back to this. I added some stuff to the wiki page. I don't know whether that covers every use-case, but I think it should cover most of them.

For OctoDash only the `GET` endpoints are needed. I've also seen that you just added the temperature offset functionality, so it would be a plus if that was also included in the `spools` response, but that would just be a nice plus :) Thanks for all the work! I think it's finally time to switch over to SpoolManager :)