OllisGit/OctoPrint-SpoolManager #148: Spool Manager Breaks OctoPrint Wizards

kantlivelong, 2021-04-16:
Hey @OllisGit,

I pushed out a new PSU Control release last night which introduced a simple message conveyed via the wizard viewmodel and had an issue where users are unable to complete/close the wizard. After some help of the users we narrowed the cause down to Spool Manager. 

SpoolManager: SpoolManager Plugin (1.3.3)

It's easy to reproduce and can be triggered on a fresh install:
1. Install Spool Manager
2. Stop OctoPrint
3. Set `firstRun` to `true` in `config.yaml`
4. Clear `seenWizards`
5. Start OctoPrint
6. Click through the wizard but get stuck on finish.

js console throws:
```
Uncaught TypeError: self.settingsViewModel.saveData() is undefined
    finishWizard http://127.0.0.1:5000/static/js/app/viewmodels/wizard.js:221
    onFinish http://127.0.0.1:5000/static/js/app/viewmodels/wizard.js:191
    jQuery 20
    show http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1747
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1809
    jQuery 2
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1805
    jQuery 20
    show http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1747
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1809
    jQuery 2
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1805
    jQuery 20
    show http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1747
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1809
    jQuery 2
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1805
    jQuery 8
wizard.js:221:18
    finishWizard http://127.0.0.1:5000/static/js/app/viewmodels/wizard.js:221
    onFinish http://127.0.0.1:5000/static/js/app/viewmodels/wizard.js:191
    jQuery 20
    show http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1747
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1809
    jQuery 2
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1805
    jQuery 20
    show http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1747
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1809
    jQuery 2
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1805
    jQuery 20
    show http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1747
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1809
    jQuery 2
    tab http://127.0.0.1:5000/static/js/lib/bootstrap/bootstrap.js:1805
    jQuery 8
```

Nothing of value appears to be on the py side of things as far as logging goes.

Will provide more info as I find it.

kantlivelong, 2021-04-16:
https://github.com/kantlivelong/OctoPrint-PSUControl/issues/181

kantlivelong, 2021-04-17:
Guess this is a dupe of #143

OllisGit, 2021-04-17:
Hi @kantlivelong,

thanks a lot for the detailed description. I was able to reproduce it and thankfully @ralmn provided a patch.

New version 1.3.4 is out now

OllisGit, 2021-04-17:
Btw. I have no idea why I overwrite the orig. function instead of using the ```onSettingsBeforeSave()``` event.

moarkaboom, 2021-04-17:
Solved the issue for me, thanks!

closed 2021-04-18 by OllisGit