OllisGit/OctoPrint-SpoolManager #149: Cooling Fan Offset

jbienz, 2021-04-18:
Similar to #61 which is temperature offset for filament, I'd also like to request part cooling fan offset. I don't know what we would assume is default (maybe 100%?) but it would likely need to be applied in a similar way. This would also require adding a cooling offset field to the spool information page.

Thanks for considering.

OllisGit, 2021-05-24:
puhh.....OP doesn't support assigning offset for the fan. Maybe you can raise a request in the OP-Issue tracker.
After OP support this I can add this feature.

jbienz, 2021-05-24:
Maybe I'm confused. I thought the way tool temperature offset worked was by looking at the temperature set by the GCODE and adding or subtracting a number from it. I know cooling fan speed can be set, but are you saying there isn't a way for an OP plugin to get notified when GCODE makes a cooling fan speed update?

OllisGit, 2021-05-24:
Hi @jbienz,

of course a plugin could listen to all gcode-lines and react on this...or even better OP provide an API where you can say: here is my offset "do the right thing"

The first part needs to be implement (UI + Server logic), testet and improved...or you choose an already proofed api.

.... and I am not sure how many people need this feature.

closed 2022-01-24 by github-actions[bot]

OllisGit, 2022-01-24:
Just for the records: When OP could provide an Fan-Offset-API I could be implemented. in SpoolMaager
OP Issue: https://github.com/OctoPrint/OctoPrint/issues/3692