OllisGit/OctoPrint-SpoolManager #150: Unable to import CSV

airbornetrooper82573, 2021-04-20:
I just installed SpoolManager and attempting to upload my csv from filament manager. It never does anything. CSV Import-Status never shows a status no matter how long I let it sit for.

Xiondewah, 2021-04-20:
I have the same issue.

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 the issue should be fixed.

Please test and give me a feedback.

Thx, in advance
Olli

airbornetrooper82573, 2021-06-10:
I updated to latest version, exported current Filament Manager and attempted to upload. Still showing no progress. Let it sit for a few minutes and restarted, still not showing database.

hgross, 2021-08-16:
Had the same issue and traced it to my reverse proxy setup.
If you guys are using a proxy in front of octoprint, check #194 - also contains a workaround for nginx users.

OllisGit, 2022-01-07:
Issue still present?

closed 2022-02-18 by github-actions[bot]