OllisGit/OctoPrint-SpoolManager #153: Multimaterialsupport like Filamentmanager

BenBetta, 2021-05-01:
It would be nice to choose the filament for all 5 positions in my mmu equipment. So on automatical or manual tool change also the counter for the filament changes to get accurate information about used filament on each spool.

codingcatgirl, 2021-05-14:
This is a duplicate of #46, which is also stated in the README ;)

disconn3ct, 2021-05-14:
#46 is closed though?

codingcatgirl, 2021-05-14:
Yeah, thanks to stalebot. But the last comment should explain what's going on?

codingcatgirl, 2021-05-15:
I was waiting for the feature for so long that i just wrote it myself :D Here's the pull request, it's fully running on my branch: #161

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 the PR from @codingcatgirl is now included. It looks really nice. Thanks again!!! 

Please test and give me a feedback.
Olli