OllisGit/OctoPrint-SpoolManager #154: Custom command sequence per filament

brownowski, 2021-05-02:
Would it be possible to incorporate custom start and end command sequences per filament that run before/after the print job?

My use case would be for setting pressure advance and extruder step distance values for different filaments before a job starts.

OllisGit, 2021-05-24:
Hi @brownowski,
nice idea, but what is  the "best location" to assign the "spool-gcode" into the print.

We have the provided gcode from the slicer and then we (could) have the gcode from OP.

Or should the "spool-gcode" injected to a specific location. E.g.
define a special comment: ```; INSERT SPOOLMANAGER-BEFORE SPOOL-CODE```
```
...
M109 S212
G92 E0
;G1 X10.0 Y10.0 F2000
;G1 Z0.1  F1000
;G1 Y60.0 E4.0 F1000          	  ; Prime the nozzle
;G1 Y150.0 E10.0 F500	  ; Prime the nozzle
;G92 E0
M117 Hot and Oozing Now!
G92 E0
G1 E-7.5000 F3600
G1 Z0.075 F1020
; process PLA
; layer 1, Z = 0.075
T0
; tool H0.100 W0.400
; INSERT SPOOLMANAGER-BEFORE SPOOL-CODE
; skirt
G1 X61.012 Y78.349 F1800
...
```

Domi1412, 2021-05-29:
I think the best was ist to handle the position by specific comment

OllisGit, 2021-05-29:
Hi @Domi1412,
I found an easy way to implement such feature: Using GCODE-Scripts in OP
https://docs.octoprint.org/en/master/features/gcode_scripts.html

In our case it looks like this:
[image]

And of course in the Spool-Dialog there are two textareas to define the gcode for each event.

What do you think?

Domi1412, 2021-05-29:
Hi

It looks good.

OllisGit, 2021-05-29:
...but there is one thing I need to solve...I need to pass thru the selected tool-number to the custom-script, so that e.g. the temp of the right spool is increased instead of all tool-heads.

Domi1412, 2021-07-23:
Hi

do you have some new information of the status of this CR?

katiekloss, 2022-12-10:
Hi! Is this something I could take a stab at, or would I be duplicating effort if it's already in progress? I have the same use cases as @brownowski