OllisGit/OctoPrint-SpoolManager #156: Spool QR code points to http instead of https

jkaessens, 2021-05-04:
The spool QR code contains `http` even if the UI is loaded via `https`. Octoprint is run through HAproxy, so flask can't possibly know about the real URL. It's probably safer pick the correct URL from the actual HTTP request header, or have a config option to override Octoprint's base URL.

OllisGit, 2021-05-19:
hmm..is there a config-setting in OctoPrint I missed?
Is OP running out of the box after adding a proxy in front of it?

OllisGit, 2021-05-19:
I found an topic in the OP-Forum:

https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107

> If you also want to run it SSL encrypted with HAProxy being the SSL terminator, you should also send a header of X-Scheme with each forwarded request to indicate https requests. You can configure this in the backend section like this:
>
>backend octoprint
> ...
> reqadd X-Scheme:\ https if { ssl_fc }
> ...

OllisGit, 2021-05-19:
fyi: I ask the community: 
https://community.octoprint.org/t/plugin-url-behind-haproxy/34276

OllisGit, 2021-05-21:
Hi @jkaessens,
I want to reproduce the issue, so I want to setup my own HAProxy.
It would be helpfull if you could provide me your haproxy-config.

jkaessens, 2021-05-31:
Actually it's a super plain haproxy stanza:

```
backend octoprint
        reqrep ^([^\ :]*)\ /(.*)     \1\ /\2
        option forwardfor
        server octoprint1 127.0.0.1:5000

backend webcam
        reqrep ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8080
```

I feel that forwarding the existing parameters via HTTP headers (like in the HAProxy forum topic) is not how it's supposed to work. After all, the end point should already know everything from existing headers. I might be wrong though, and try the suggested workaround to see if it also works for SpoolManager. I'll check back with you on that.

OllisGit, 2021-06-04:
Hi @jkaessens,
I setup a local HA-Proxy and I could reproduce the issue. 
With the default ha-proxy config it returns always http instead of https.
You need to add the header ``` X-Scheme:https``` (as described here: https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107)

I used a newer version of haproxy, so the the statement looks like this (instead of ```reqadd X-Scheme:\ https if { ssl_fc }```)
```
backend octoprint
        http-request add-header X-Scheme https if { ssl_fc }
        option forwardfor
        server octoprint1 127.0.0.1:5000
```

normen, 2021-07-24:
I seem to have the opposite issue, I get `https` in the QR code while my OctoPi instance isn't even reachable through https, running behind haproxy with only `http` port `80` forwarded. Using the exact same config as posted above by jkaessens.