OllisGit/OctoPrint-SpoolManager #158: SpoolManager für mehrere Drucker

askabbs, 2021-05-12:
Moin,
ich würde gern den SpoolManager in Verbindung mit CostEstimation nutzen. Jedoch sind die Programme für OctoPrint mit nur einem Drucker ausgelegt. Ist evtl. eine Version im MultiPrinterMode geplant ? Ich dachte so an ein Background-Programm, bei dem ich mit der entsprechenden Session als aktiv anmelden kann und das beim QR-Code-Scannen aufgerufen wird, die Daten an die angemeldete Session weiter leitet und anschliessend das Aktiv-Flag löscht. So wäre eine zentrale Filament-Verwaltung möglich.
Ich habe z.B. das Upload-Verzeichnis auf einen Windows-Rechner gelegt und so alle gcode-Dateien zentral liegen auf die ich mit allen Duckern (OctoPrint-Session's) zugreifen kann. So erspare ich mir das Uploaden in 3 Verzeichnisse. 

Nur mal so als Idee.

codingcatgirl, 2021-05-14:
Eine externe Datenbank würde das Problem lösen, oder? Laut README ist das bereits in Planung. Ich könnte mir außerdem Vorstellen, dass es aus Respekt vor internationalen Mitlesenden ganz schön wäre, in diesem Englischsprachigen Repo Tickets auch auf Englisch anzulegen :)

English Translation: An external databse would solve this problem, right? According to the README that's in the works already.

codingcatgirl, 2021-05-14:
Duplikat von #126 und #5

OllisGit, 2021-05-24:
Hi @askabbs,
fyi: Latest Version of CostEstimation works with the latest SpoolManager plugin.

I already prepared some things in the sourcecode to provide a shared database. The implementation of that feature an even more the testing takes a lot of time....but it is the next big-thing ;-)

closed 2022-01-07 by OllisGit