OllisGit/OctoPrint-SpoolManager #159: SpoolManager gets broken by CalibrationCompanion

GylRon1, 2021-05-13:
If CalibrationCompanion is installed and active, the add spool pop-up window is empty. I issued this too at the github page of CalibrationCompanion. Attached an empty window and the regular window.
[image]
[image]

tgkgate, 2021-05-14:
Can confirm,  window is broken with cal companion installed

GylRon1, 2021-05-14:
iFrostizz is looking in to it. You can follow here:
https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/issues/7#issuecomment-841132680

akrilov, 2021-05-14:
Can confirm on my side as well. 
After installing calibration companion spool manager shows empty edit spool page.

GylRon1, 2021-05-17:
Bug fixed by iFrostizz: https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/releases

closed 2021-05-17 by GylRon1