OllisGit/OctoPrint-SpoolManager #160: Export to Filament Manager doesn't work.

codingcatgirl, 2021-05-14:
There seems to be an export for FilamentManager. However, the output format is a CSV, which is not at all what Filament Manager supports, it needs a ZIP file with two CSVs inside. How do i migrate back to FilamentManager? I need multi-tool support.

OllisGit, 2021-05-19:
Hi @codingcatgirl,

there is no "backport-feature" available. Only from FM to SPM.

codingcatgirl, 2021-05-20:
Oh, so the feature there is to export the current FilamentManager data in the SpoolManager format? That would make more sense, but is definitely a bit confusing / easy to misunderstand :D (To clarify, i'm talking about this feature: https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/octoprint_SpoolManager/templates/SpoolManager_settings.jinja2#L131

closed 2022-01-07 by OllisGit