OllisGit/OctoPrint-SpoolManager #162: Cant add any Spools, form to enter values is blocked by something

fenpaws, 2021-05-15:
Heya, yeah sorry for that wired title but I have a strange issue.

## Systeminfo (OctoPrint)
[octoprint-systeminfo-20210515215315.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/6488196/octoprint-systeminfo-20210515215315.zip)

## General description

When I want to add any Spools I have I only get a screen like this, it seems that something is blocking the form.
[image]

## What I tried/found out

I disabled every addon that would change the UI of OcoPrint, in my case OctoPrint-WideScreen and Themeify, nothing worked. 
After that tried a bit more and I found out after I delete this div it can see the form, but without any buttons to add it
[image]

[image]

cjmalchow, 2021-05-16:
I am also having this same issue

fenpaws, 2021-05-16:
@cjmalchow Yeah, I can confirm that. Somehow these two addons are incompatible.

OllisGit, 2021-05-24:
Hi,
in the latest release V1.4.0 I did some UI-Improvements.

Please test and give me a feedback.

Thx,, in advance
Olli

Xiondewah, 2021-05-26:
I've updated to the latest release but still have this issue.  I have disabled calibration companion and the issue resolves.