OllisGit/OctoPrint-SpoolManager #164: 1.4.0 Lost all old DB after update and can't add new ones.

SM-26, 2021-05-24:
After the new update
I've lost all of my old data, and now when I try to add a new spool
I'm getting this error
[image]

Here is what I've tried to enter
[image]

In addition, when I open the material tab, it's coming blank
[image]

I've also tried to reset all of the setting to no avail
[image]

Full log is here
[octoprint (7).log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/6534552/octoprint.7.log)

This bug stops me from printing!
so I must disable the plugin for now.

If there is any more info that I could fill in, just tell me how can I help.

OllisGit, 2021-05-24:
Hi @SM-26,

there is something wrong with the database migration-order. Could you please provide me your database ?
[image]

OllisGit, 2021-05-24:
Hi @SM-26,

I created a bug-fix version 1.4.1. Please test and give feedback.

Thx, in advance
Olli

SM-26, 2021-05-25:
Yep, 1.4.1 fixed it.
and like you said, all of my all data is back.

Do you still need the DB file?
[spoolmanager.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/6537532/spoolmanager.zip)

In any case, Thank you for you hasted assistance.
one last thing before I close this issue,

a small UI "bug" I found
[image]
[image]

We are missing the H in Hide.

closed 2021-05-25 by SM-26

OllisGit, 2021-05-25:
Thx, for the response. The 'h' bug will be fixed in next release.