OllisGit/OctoPrint-SpoolManager #165: Spools UI degraded after 1.4 update

jbienz, 2021-05-24:
The Spools UI seems degraded after the 1.4 update. Lots of empty space on the sides and toward the bottom.

[image]

OllisGit, 2021-05-24:
Hi @jbienz,

what screen resolution do you use and what browser, what zoom-factor?
In the past some other plugins "collide" with spool-manager, please disable other plugins to see if it helps
E.g. I am using chrome and everything looks fine.
[image]

achampion, 2021-05-25:
I'm using chrome and have a similar issue, they fields do not layout in 2 columns:
[image]

OllisGit, 2021-05-25:
Closed, duplicated! Tracked now here: #142 , because plugin author of "UI Customizer plugin" is involved.

closed 2021-05-25 by OllisGit