OllisGit/OctoPrint-SpoolManager #168: Can not print without a spool selected in 1.4.0

The-EG, 2021-05-25:
With the new version, 1.4.0, I can no longer start a print with no spool selected. 

Steps to reproduce:
1. Un-select a spool in the spool manager pane:
    [image]
2. Select a file and click 'Print'

The UI doesn't respond in anyway, but this error does occur in the javascript console:
```
Uncaught TypeError: result is undefined
    confirmSpoolSelectionBeforeStartPrint http://10.0.0.102/ender-3-v2/static/webassets/packed_plugins.js?87fa31a1:4470
    allowedToPrint http://10.0.0.102/ender-3-v2/static/webassets/packed_plugins.js?87fa31a1:4998
    c http://10.0.0.102/ender-3-v2/static/webassets/packed_libs.js?86afb3b9:13
    fireWith http://10.0.0.102/ender-3-v2/static/webassets/packed_libs.js?86afb3b9:13
    l http://10.0.0.102/ender-3-v2/static/webassets/packed_libs.js?86afb3b9:13
    o http://10.0.0.102/ender-3-v2/static/webassets/packed_libs.js?86afb3b9:13
    sentryWrapped http://10.0.0.102/ender-3-v2/static/webassets/packed_core.js?7846f86e:634
```

Prior versions showed a confirmation dialog, and started the print if the appropriate response was chosen.

OctoPrint 1.6.1, FireFox 88.0.1

OllisGit, 2021-05-25:
Hi @The-EG,

issue should be solved in latest release 1.4.3.

Please test and give feedback.

Thx, in advance
Olli

The-EG, 2021-05-25:
Confirmed fix in 1.4.3.  Thanks!

closed 2021-05-25 by The-EG