OllisGit/OctoPrint-SpoolManager #169: Show cost estimation

wernight, 2021-05-25:
We already have all the info, it's just a matter of showing the cost in the UI.

There is PR for an old plugin https://github.com/malnvenshorn/OctoPrint-CostEstimation/pull/30 but there is no action done there. It might be just easier and nicer to merge that plugin with SpoolManager.

Note: Yes Cura has a cost estimator as well which is useful. Having the cost in OctoPrint has the advantage that a single database could be used and power usage could be factored in (it adds about 15% to the cost for me but it's based on print time and temperature mostly).

codingcatgirl, 2021-05-25:
…uh, this should already work. I wonder if something broke during the mutli-material release the other day?

codingcatgirl, 2021-05-25:
Oh, i see your problem. You're looking at the old version of the CostEstimation repo. The repo actually has been taken over be the same maintainer as spoolmanager. You can find it here and it does support it. Octoprint also should even tell you that you are on the wrong fork. https://github.com/OllisGit/OctoPrint-CostEstimation

wernight, 2021-07-07:
Awesome. Feel free to close or leave open as a FAQ.