OllisGit/OctoPrint-SpoolManager #172: Odd display issue, grey box over everything

moorsey, 2021-05-26:
I have just moved over from Filament Manager, did a CSV import, which worked eventually

However, trying to update colours etc, but am just seeing this when I click on any spool

[image]

I can tab through the fields and see some fields, which looks like this:

[image]

The Add Spool option displays the same way

OllisGit, 2021-05-26:
Hi @moorsey,

do you use the plugin UI Customizer?

moorsey, 2021-05-26:
Hi @OllisGit 

Sorry, should have mentioned, no I don't, did see that mentioned as a problem app when I was looking through the issues logged here

OllisGit, 2021-05-27:
Hi @moorsey,

- which version of OP do you use?
- is the problem in all browsers (chrome, firefox, ...)?
- which screenresoution do you use?
- when you change the zoom-level in the browser it the dialog then  visible?

moorsey, 2021-05-29:
Morning

- Octoprint - 1.6.1
- Same in Chrome and Firefox
- Have two screens, 3440x1440 and 2560x1440, same on both.  Did try re-sizing browser window around a bit, same at all sizes
- Moved between 25 and 500% zoom, stayed the same at all zoom levels on both screens

Do have other screens I can try stuff on, will get a 1920x1080 hooked up later

Sivv3k, 2021-05-31:
Hello, Exactly same screen/frame issue :/

powers, 2021-05-31:
Same Issue. 
Octoprint - 1.6.1
Same in Chrome, Opera, and Firefox
Monitor: 1920x1080
Zoom level doesn't affect it.

farfalleflickan, 2021-06-05:
Same issue here, same specs/setup (no UI customizer) too.

I noticed that the `class=modal-body` div has also the `hack-flowbreak` class (from `packed_plugin.css`). This sets `overflow: visible !important`, if that is removed then it behaves normally.

dwery, 2021-06-13:
Same, I can confirm @farfalleflickan workaround works for me

mtabo, 2021-06-16:
> Same, I can confirm @farfalleflickan workaround works for me

Same here, @farfalleflickan workaround works for me too

moorsey, 2021-06-17:
For someone not so familiar, where would I find the `packed_plugin.css` file to make the edit suggested by @farfalleflickan ?

Had a poke around in the filesystem, but not seeing anything too obvious!

r748, 2021-06-21:
Same issue for me as well. Workaround by @farfalleflickan resolves the problem.

k0rls8, 2021-06-23:
I was having the same issue but it has cleared up. Unable to explain why unless it was a windows update that made the difference. Running Octoprint 1.6.1.