OllisGit/OctoPrint-SpoolManager #173: Description of CSV import fields

moorsey, 2021-05-26:
Is there a reference anywhere for the import fields and what they mean?

For example, what is the difference between "Total Weight" and "Spool Weight"?

As I am moving from Filament manager, I am trying to figure the minimum that I need to enter for everything to work, do I need lengths as well as weights?  Do I need Density?

These are all fields above what Filament Manager uses, but don't know what gaps need filling

OllisGit, 2021-05-26:
Hi @moorsey,

currently there is no description of these fields...., but you can start with a description and any field wich is unclear I can add my description (create a wiki-page or add the description here)

Did you saw the download link of the sample csv-file?
 
Spool Weight= mostly "black" plastic roll
Total Weight  = Filament + Spool Weight

moorsey, 2021-05-26:
Thanks for the info @OllisGit 

Yes, the sample CSV was what I used to move info across from Filament Manager

Right, that makes sense on the weights, I had but my data in Spool Weight and left Total Weight blank.  Have moved this across and will re-import

I think I'm OK with the rest, just not sure if I needed to complete the Density field?

Much appreciated

OllisGit, 2021-05-28:
[image]

closed 2021-05-28 by OllisGit

moorsey, 2021-05-29:
Gotcha, missed that, apologies, many thanks

6ffm70, 2021-12-26:
fyi, I have created a wiki page that allows putting comments to single attributes [https://github.com/OllisGit/OctoPrint-SpoolManager/wiki/Spool-Attributes](url)