OllisGit/OctoPrint-SpoolManager #175: Confirm dialog sometimes doesn't pop up.

codingcatgirl, 2021-05-26:
As far as i can tell, this is how you should be able to reproduce it: Have a spool already selected and be connected to the printer, then connect to the octoprint interface. Select a print, klick Print → the dialogue will not pop up. Dis/reconnect from/to the printer, klick the print button again, and the confirm dialog will pop up. Weird?

cicinovec, 2021-05-26:
I confirm, after update to 1.4.1 the confirmation pop-up window appeared for the first time, but now not anymore

OllisGit, 2021-05-26:
Sorry,
I can not reproduce the issue. What I did:

- Open a fresh browser-tab 
- Load the OP-Interface 
- Printer is autoconnect and no spool is selected
- Select a spool
- Close browser tab
- Load the OP-Interface
- Select a file
- Click Print
- Popup appears

killernils, 2021-05-27:
Just updated to 1.4.3 and stopped getting confirmations/popups on:
"Warn, if no spool is selected."
"Reminder for verifying the selected spool."

Anything I can do to help out?

cicinovec, 2021-05-27:
After updating to 1.4.3 is the dialog back

gn0m3r4l, 2021-06-02:
Same for me.
Plugin Version 1.4.3
Octoprint 1.6.1

Unfortunately, I cannot tell when this occurs first time.

ales444, 2021-06-02:
Same for me.
Plugin Version 1.4.3
Octoprint 1.6.1

Stopped working recently, not sure exactly when.

codingcatgirl, 2021-06-05:
I just got thisagain for the first time in days. Still no idea what caused it. Could be a printer that i aborted and then restartet, maybe?

codingcatgirl, 2021-06-15:
Sorry to say tha this is still happening.

OllisGit, 2021-06-17:
I have an idea.....the current implementation reads the needed (calculated) filament length for each used tool from the metadata-file (.metadata.json) located in the upload folder.
[image]

The implementation expect that this data is available after selecting the file for printing.
But this length is calculated depending on the selected strategy in the OP-Settings (Settings / Features).
[image]

Because at the moment it it just an idea, but you can also test it, if you change to always and/or you can select the file, select an other file and come back to the file....in the meantime the length should be calculated and the popup should appear......a little bit complicated.

I think I will write a patch for that ;-). ....after I can confirm the idea!

codingcatgirl, 2021-06-18:
Oh, yeah, that sounds like it could be it! Though it might be even simpler… the analysis could simply take a while and not be done yet and that's enough to trigger the bug? The solution would probably be to just assume all slots are used if the analysis isn't done yet

MrRexRuthless, 2021-07-19:
I am having the same issue.  I have to deselect a spool to get it to print.

closed 2021-11-03 by OllisGit